realsenseai/librealsense #3013: RGBD camera depth image registration, there are no corresponding cases Python？

honghande, 2019-01-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseCustomerSupport, 2019-01-21:
Hi honghande,

I meant to ask you please provide more details in Description section of the question/issue that you are reporting here. This helps us better support.

Thanks!

honghande, 2019-01-22:
hi,@RealSenseCustom
OK, I mean: 1. Without align to function for registration of depth camera and RGB, I want to use D405 to collect depth image and color image for registration, to generate depth image with color, or to generate point cloud with color;

2. My camera only uses two infrared cameras, one with color. Can I do RGBD registration? Is there a demo to refer to?

chaojizhangdaxin, 2019-01-25:
self.align_to = rs.stream.color
self.align = rs.align(self.align_to)
self.frames = self.pipeline.wait_for_frames()
self.aligned_frames= self.align.process(self.frames)
这四行代码是让RGB照片和深度照片对齐

RealSenseCustomerSupport, 2019-01-25:
Hi honghande,

Guess you have checked out this sample: https://github.com/IntelRealSense/librealsense/tree/master/examples/align
Does it help or you have other questions?

Thanks,
Richard

RealSenseCustomerSupport, 2019-02-04:
----------------------------------------------

Hi honghande,

If there is nothing else, this one will be closed.

Thanks!

closed 2019-02-19 by RealSenseCustomerSupport