realsenseai/librealsense #3014: Is align_to what we call depth image and color image registration?

honghande, 2019-01-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-01-07:
Yes. [Usage example](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb).

honghande, 2019-01-08:
Hi，@dorodnic ，
I want to achieve the registration of depth image and RGB image, but my binocular camera does not have RGB camera, only two infrared cameras, one with color, can we use color infrared instead of color camera? My code is as follows: Is there a mistake in writing this way?
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)

pipeline.start(config)
profile = pipeline.get_active_profile()
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

def alignmentDepthandColor(depth_frame,color_frame):
    # Convert images to numpy arrays
    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    colors = []
    depth_video_profile=depth_frame.profile.as_video_stream_profile()
    color_video_profile=color_frame.profile.as_video_stream_profile()
    depth_intrin = depth_video_profile.intrinsics
    color_intrin = color_video_profile.intrinsics
    depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
    color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)
    print("\n Depthintrinsics：{}".format(str(depth_intrin)))
    print("\n Colorintrinsics：{}".format(str(color_intrin)))
    print("\n Depthtocolorextrinsics：{}".format(str(depth_to_color_extrin)))
    if len(depth_image[0][0]) == 3:
        depth = cv2.cvtColor(depth_image, cv2.COLOR_BGR2GRAY)
    rows = depth.shape[0]
    cols = depth.shape[1]

    for row in range(0,rows):
        for col in range(0,cols):
            # depth_value = depth_frame.get_distance(row, col)
            depth_value=depth[row][col]* depth_scale
            if depth_value == 0:
                pass
            depth_pixel=[row,col]
            print("\n\t depth_pixel:0},value：{0}meter".format(str(depth_pixel),str(depth_value)))
            depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin,depth_pixel,depth_value)
            print("\n\t 3D depth_point:{} ".format(str(depth_point)))
            color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin,depth_point)
            print("\n\t 3D color_point:{}".format(str(color_point)))
            color_pixel = rs.rs2_project_point_to_pixel(color_intrin,color_point)
            print("\n\t color_pixel:{} ".format(str(color_pixel)))
            colors.append(color_pixel)
    return colors
# Streaming loop
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue
        depth_color = alignmentDepthandColor(depth_frame, color_frame)
        # Render images
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', depth_color)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()

honghande, 2019-01-08:
Hi，@dorodnic ，I got two infrared cameras with the following parameters:
rotation: [1, 0, 0, 0, 1, 0, 0, 0, 1]
translation: [0, 0, 0]

RealSenseCustomerSupport, 2019-01-18:
Hi Honghande,

You can use the "left" IR camera color frame which is overlapped, pixel aligned, and sync'ed with the depth frame. Hence no color-depth alignment is needed.
To enable Left IR to stream color:
`config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 30) //may use rgb8 format on D410/415`

closed 2019-02-12 by RealSenseCustomerSupport