realsenseai/librealsense #3018: Frames streaming blocked in C# wrapper

walterbaccinelli, 2019-01-07:
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |   2.17.0}                          | 
| Language                            |  C#                          |  

### Issue Description
I've been using the C# wrapper for developing a software that performed the streaming of RGB and depth images. Everything worked fine with SDK version 2.16.0, but changing to 2.17.0 produced some issues. In fact, I am able to get some frames correctly and then the camera continuously streams the same frame. I correctly release all the resources and the objects, so no problem is present in that side. Moreover I can see that the camera is still streaming new images, but the FrameSet is always the same. Unplugging and plugging again the device resume the correct streaming for some frames, that the same behavior as before is detected.
Do you have any idea on how to solve this issue?

Thank you,
Walter

amatabuena, 2019-01-11:
Hi, 

I don't know if your issue is the same, but maybe it's the same one that I reported three days ago and "resolved" today (making a lot of tests). This is my issue:
https://github.com/IntelRealSense/librealsense/issues/3026#issue-397010477

I hope it can help you.

jb455, 2019-01-11:
@walterbaccinelli, are you streaming the highest resolution depth and colour streams? I've seen the colour image 'sticking' before (updating every few seconds instead of ~30 times a second, depth remains at ~30fps) with that configuration, I assumed it was a USB bandwidth issue. Turning depth resolution down to 848*480 was an effective workaround for me, though it would be nice to be able to use the 1280*720 depth stream.

RealSenseCustomerSupport, 2019-01-25:
@walterbaccinelli  Any update about this issue? Looking forward to your update. Thanks!

walterbaccinelli, 2019-01-25:
> Hi,
> 
> I don't know if your issue is the same, but maybe it's the same one that I reported three days ago and "resolved" today (making a lot of tests). This is my issue:
> [#3026 (comment)](https://github.com/IntelRealSense/librealsense/issues/3026#issue-397010477)
> 
> I hope it can help you.

walterbaccinelli, 2019-01-25:
Thank you @amatabuena, my issue was indeed very similar to yours. coping the frame before doing any operation seems to prevent the streams from get blocked.

closed 2019-01-25 by walterbaccinelli