realsenseai/librealsense #3019: Installing SDK 2.0 on Win 10 Python 3.6

MaryBD, 2019-01-07:
Hi, I am trying to install pyrealsense2 on Pycharm to use Intel RealSense D435. On the website there is a Python wrapper but as mentioned: "This wrapper does not support newer versions and does not work with the RealSense SDK 2.0.". Is there any wrapper for Python 3.6? I already used the wrapper for Python 2.7 and it works perfectly but I need to get it work with Python 3. Any advices? 

Camera Model IntelRealSense D435
Operating System & Version | Windows 10
Platform | Laptop
SDK Version | 2.0

Many thanks
Kind Regards

dorodnic, 2019-01-08:
[pyrealsense**2**](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md) does support D435 and SDK 2.0. Perhaps you are referring to pyrealsense.
`pip install pyrealsense2`

MaryBD, 2019-01-10:
I have already installed it. I tried to use python 3.6 instead and pyrealsense2 works perfectly in this platform. Thanks.

closed 2019-01-10 by MaryBD