realsenseai/librealsense #3020: Record .bag file containing miscalenous topics (not only RealSense)

MTDzi, 2019-01-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |  Ubuntu 16 / Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-43  /     4.4.38-tegra                                   | 
| Platform                                 | PC / NVIDIA Jetson TX1 |
| SDK Version                            | 2.17.1                        | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description

I would like to record a ROS .bag file on my Jetson TX1, capturing data from two topics: `/camera/depth/image_rect_raw` and `/joy`. Then, I would like to read the .bag file, use some combination of post-processing functions, and serialize the data (doesn't matter how, but let's say: to TFRecord).

I produced a .bag file using `rosbag record -O whatever --lz4 /joy /camera/depth/image_rect_raw`, but when I used [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py) script, I learned that:
```
  File "rs_extract_frames.py", line 47, in <module>
    pipeline.start(config)
RuntimeError: Failed to resolve request. Request to enable_device_from_file("house/bags_2019-01-06-14-33-36.bag") was invalid, Reason: Failed to create ros reader: Invalid file format, file does not contain topic "/file_version" nor "/FILE_VERSION"
```

So, **my question is**: how should I record a .bag file containing all the necessary data for later post-processing experiments?

dorodnic, 2019-01-08:
SDK ROS-bag is (mostly) valid ROS-bag, but not every ROS-bag can be opened by the SDK. 
[src/media/readme.md](https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md)
[doc/sample-data.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md)
There are ways to record external data in SDK ROS-bag format, but for what you are trying to do, I'd recommend to use ROS alternatives. librealsense is only designed to handle camera data

MTDzi, 2019-01-08:
OK,

Now I know for sure, thanks!

closed 2019-01-08 by MTDzi