realsenseai/librealsense #3021: librealsense - build from source - librealsense solution got red flag beside each project

ShilongLIU, 2019-01-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-01-08:
The red icon is not related to the error. 
If you just need precompiled version of the SDK you can download binary installer from:
https://github.com/IntelRealSense/librealsense/releases
The error is not strictly related to RealSense, but can occur when compiling C++ code. 
There are several proposed solutions [on Google](https://www.bing.com/search?q=atlstr.h%20not%20found&cbir=sbi&imageBin=&qs=n&form=QBRE&sp=-1&pq=atlstr.h%20not%20fou&sc=1-16&sk=&cvid=8D391DD7789442D09A1A0ED888433807), from adding ATL support in VS installation to installing Windows SDK

ShilongLIU, 2019-01-08:
Thanks very much Dorodnic.
It works for me.
I have added the atl support via vs installer.

closed 2019-01-08 by ShilongLIU

Alex-OPTIM, 2020-05-08:
During `npm install node-librealsense` I had a few errors where one of them was: `Cannot open include file: 'atlstr.h'`.
Adding ATL support in VS installer helped me too. Thanks @dorodnic