realsenseai/librealsense #3022: realsense-viewer doesn't work on realsense D435

jinfagang, 2019-01-08:
I just installed libsense sdk and used to success load and show D410, but convert to D435, everything can not work, both realsense-viewer and ros node. Here is the error:

```
[ INFO] [1546932234.573819335]: ROS Node Namespace: camera
[ INFO] [1546932234.573841668]: Device Name: Intel RealSense USB2
[ INFO] [1546932234.573855767]: Device Serial No: 829212070312
[ INFO] [1546932234.573867564]: Device FW version: 05.09.02.00
[ INFO] [1546932234.573880090]: Device Product ID: 0x0AD6
[ INFO] [1546932234.573892396]: Enable PointCloud: On
[ INFO] [1546932234.573904604]: Align Depth: Off
[ INFO] [1546932234.573915532]: Sync Mode: On
[ INFO] [1546932234.574118139]: Device Sensors: 
[ INFO] [1546932234.574228273]: Stereo Module was found.
[ INFO] [1546932234.574351865]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546932234.574374447]: setupPublishers...
[ INFO] [1546932234.576676210]: Expected frequency for depth = 30.00000
[ INFO] [1546932234.597895728]: setupStreams...
 08/01 15:23:54,598 WARNING [140675837327232] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ WARN] [1546932234.599696703]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: Z16, Width: 640, Height: 480, FPS: 30
[ INFO] [1546932234.599911518]: Add Filter: pointcloud
[ INFO] [1546932234.603156352]: num_filters: 1
[ERROR] [1546932234.603268486]: Given depth profile is not supported by current device!
[FATAL] [1546932234.816165024]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-1] process has died [pid 14604, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jintain/.ros/log/cc7ae830-12f2-11e9-9e62-3497f693b553/camera-realsense2_camera_manager-1.log].
log file: /home/jintain/.ros/log/cc7ae830-12f2-11e9-9e62-3497f693b553/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has died [pid 14605, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/jintain/.ros/log/cc7ae830-12f2-11e9-9e62-3497f693b553/camera-realsense2_camera-2.log].
log file: /home/jintain/.ros/log/cc7ae830-12f2-11e9-9e62-3497f693b553/camera-realsense2_camera-2*.log

```

Why is the error got? I have already rebuilt the librealsense sdk

jinfagang, 2019-01-09:
@dorodnic  Well, realsense-viewer does have log but just some pop up window said stream out of lenght etc. So I using realsense ros to test if the node can not.

Just fail, my machine used to successfully load D415, but not work for D435

dorodnic, 2019-01-09:
Hi @jinfagang 
`rs-enumerate-devices` can provide some useful information.
The core of the issue, the way I see it, is that ROS node does not support D435 with USB2. 
While some camera functionality is [officially support in USB2 mode](https://realsense.intel.com/usb2-support-for-intel-realsense-depth-cameras/), not all camera capabilities can be implemented without USB3. We could update the ROS node to offer **limited** functionality with USB2, but it would be better if you could switch to USB3 for your project.

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

closed 2019-02-07 by RealSenseCustomerSupport

ebimor, 2019-03-29:
I updated the firmware to 5.11.1 and it worked.