realsenseai/librealsense #3024: USE_SYSTEM_LIBUSB does not exists in lrs_options.cmake

oiramario, 2019-01-08:
ATT

dorodnic, 2019-01-09:
Updated

oiramario, 2019-01-10:
Thanks~ Glad to see your update!
However, there are 30 macros in Build-Configuration and 26 macros in lrs_options.cmake.

Build-Configuration                                           lrs_options.cmake
BUILD_PYTHON_MATLAB                                  mismatch  BUILD_MATLAB_BINDINGS
DOTNET_VERSION_EXAMPLES                          missing
DOTNET_VERSION_LIBRARY                              missing
BUILD_CV_KINFU_EXAMPLES                             missing
INTERNAL_FW_SOURCE                                     missing
USE_SYSTEM_LIBUSB                                          missing
CMAKE_BUILD_TYPE                                           ignore

dorodnic, 2019-01-10:
Yes, we should try to collect everything into `lrs_options.cmake`. 
Recent Cmake refactor was a big step forward IMHO, and I certainly don't want to go back to the spaghetti Cmake we used to have until recently.

oiramario, 2019-01-10:
(｡•̀ᴗ-)✧

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

claudiofantacci, 2019-03-22:
I'm also interested in having `USE_SYSTEM_LIBUSB` as `ExternalProject_Add(libusb ...)` custom target sometimes fails on windowsusing `Ninja` build with
```
fatal: could not create work tree dir 'libusb': Permission denied
fatal: could not create work tree dir 'libusb': Permission denied
fatal: could not create work tree dir 'libusb': Permission denied
-- Had to git clone more than once:
          3 times.
CMake Error at C:/Users/cfantacci/GitHub/librealsense/buildprova/libusb-prefix/tmp/libusb-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/libusb/libusb.git'
```

To workaround this problem you just need to run `ninja` twice. I tried to had a look into it, but found no solution.

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

@dorodnic Do we have any plan to include USE_SYSTEM_LIBUSB option?

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@oiramario Could you please check the latest SDK v2.26? lrs_options.cmake and the related wiki doc are updated. Looking forward to your feedback. Thanks!

RealSenseCustomerSupport, 2019-09-02:
----------------------------------------------

@oiramario If no other questions, will close this accordingly. Thanks!

closed 2019-09-02 by RealSenseCustomerSupport