realsenseai/librealsense #3026: Color Frame hangs when converting to Bitmap

amatabuena, 2019-01-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 05.10.06                                                  | 
| Operating System & Version |   Windows 10                                          | 
| Kernel Version (Linux Only)   |  ---                                                          | 
| Platform                                |  x86                                                          |
| SDK Version                          |  2.17.0                                                    | 
| Language                              |  C#                                                         | 
| Segment			|  PC                                                                 | 

### Issue Description
When converting Color VideoFrame to Bitmap image gets freezes after several bitmaps. It worked fine with a previous version (2.11.0). Colorized Depth images works fine. It happends to me with with any format. I also tested it saving bitmap to disk (commented line) and the image hangs also.

This is the capture frame code:

`
public void StartCapture()
{

            try
            { 
                pipeline = new Pipeline();
                colorizer = new Colorizer();
                var cfg = new Config();
                cfg.EnableStream(Stream.Depth, 640, 360, Format.Z16, 30);
                cfg.EnableStream(Stream.Color, 640, 360, Format.Rgb8, 30);
                var profile = pipeline.Start(cfg);
                var token = tokenSource.Token;

                var t = Task.Factory.StartNew(() =>
                {
                    try
                    {
                        while (!token.IsCancellationRequested)
                        {

                            try
                            {
                                using (var frames = pipeline.WaitForFrames())
                                {
                                    var colorFrame = frames.ColorFrame.DisposeWith(frames);
                                    Bitmap bmpColor = ToBitmap(frames.ColorFrame);
                                    //bmpColor.Save(@"C:\tmp\color.bmp");
                                    if (NewFrameReceived != null)
                                        NewFrameReceived(bmpColor);

                                    bmpColor.Dispose();
                                    bmpColor = null;

                                    colorFrame.Dispose();
                                }
                            }
                            catch (Exception ex)
                            {
                                Logger.LogException(LogLevel.Error, "StartCaptureTask", ex);
                            }
                        }
                    }
                    catch (Exception e)
                    {

                    }
                }, token);
            }
            catch (Exception ex)
            {
                Logger.LogException(LogLevel.Error, "StartCapture", ex);
            }
        }

`

And this is the conversion from VideoFrame to Bitmap:
`public static Bitmap ToBitmap(VideoFrame frame)
        {
            
            var bytes = new byte[frame.Stride * frame.Height];
            frame.CopyTo<byte>(bytes);
           
            var bpp = frame.BitsPerPixel; 
            PixelFormat pf = PixelFormat.Format24bppRgb;
            switch (bpp)
            {
                case 16:
                    pf = PixelFormat.Format16bppGrayScale;
                    break;
                case 24:
                    pf = PixelFormat.Format24bppRgb;
                    break;
                case 32:
                    pf = PixelFormat.Format32bppArgb;
                    break;
            }

            var bs = new Bitmap(frame.Width, frame.Height, pf);
            var BoundsRect = new Rectangle(0, 0, frame.Width, frame.Height);
            BitmapData bmpData = bs.LockBits(BoundsRect,ImageLockMode.WriteOnly,pf);
            
            Marshal.Copy(bytes, 0, bmpData.Scan0, frame.Stride * frame.Height);
            bs.UnlockBits(bmpData);
            
            return bs;

        }`

As I said before, this code worked fine with 2.11.0 version (only including requiered changes). It also works if I do not convert to bitmap.
Any suggestions?

amatabuena, 2019-01-11:
Hi,

I respond myself. Although in 2.11.0 version the above cde worked fine, it looks like in 2.17.0 version is required a copy of VideoFrame to make it work.

I mean, this line fails:
`Bitmap bmpColor = ToBitmap(frames.ColorFrame);`

But this one works ok:
`VideoFrame colorFrame = frames.ColorFrame.DisposeWith(frames) as VideoFrame;
Bitmap colorBitmap = Utils.toBitmap(colorFrame);`

It seems a copy is requiered for each transformation or operation you make. Otherwise frame hangs.

closed 2019-01-11 by amatabuena

jinxzzz, 2019-03-06:
Do you still have problems with this code?In addition, can depth images be converted using the function ToBitmap?

tralladas, 2019-03-06:
Hi,

My problem was resolved making a copy previusly to converting to bitmap (as I explain above)

Depth frame can be also converted if you colorize It previusly.

jinxzzz, 2019-03-06:
Thank you for your quick reply. I want to know what Utils means in your code.

VideoFrame colorFrame = frames.ColorFrame.DisposeWith(frames) as VideoFrame; 
Bitmap colorBitmap = Utils.toBitmap(colorFrame);

amatabuena, 2019-03-07:
Hi, 

Utils is a class created by me containing some methods. toBitmap method is like this:

`public static Bitmap toBitmap(VideoFrame frame)
        {
            return System.Drawing.Bitmap(frame.Width, frame.Height, frame.Stride, PixelFormat.Format24bppRgb, frame.Data);
        }`