realsenseai/librealsense #3027: dfu

shaunlamont66, 2019-01-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

shaunlamont66, 2019-01-08:
Intel DFU disaster...i get a D435...old firmware, so i download the intel firmware upgrade links...and dfu...and it doesnt work!!
So, i figure its my laptop..i go buy a Windows tablet, Pentium AD i5...just in case,,,and it FAILS On both those aslo...
So...thanks for fucking nothing Intel, piece of shit, selling cameras that are bricks when they arrive
Useless bastards

HippoEug, 2019-01-09:
Yo yo yo... maybe you should describe what you've done and the errors that's been happening my friend.

dorodnic, 2019-01-09:
The issue is being closed due to severe language used, unacceptable for this forum. 
Please contact [support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) regarding the problem you are facing.

closed 2019-01-09 by dorodnic