realsenseai/librealsense #3028: when I ran box_dimensioner_multicam__demo.py, I got an error？

honghande, 2019-01-09:
My binocular camera only has two infrared cameras, no color cameras, but one of the infrared cameras is with color. How can I modify such hardware in the program?
I have revised it：
rs_config = rs.config()
		rs_config.enable_stream(rs.stream.depth, resolution_width, resolution_height, rs.format.z16, frame_rate)
		rs_config.enable_stream(rs.stream.infrared, 1, resolution_width, resolution_height, rs.format.y8, frame_rate)

1 devices have been found
Place the chessboard on the plane where the object needs to be detected..
RMS error for calibration with device number 828212141104 is : 0.0047942758443 m
Calibration completed... 
Place the box in the field of view of the devices...
Traceback (most recent call last):
  File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 158, in <module>
    run_demo()
  File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 127, in run_demo
    extrinsics_devices = device_manager.get_depth_to_color_extrinsics(frames)
  File "E:\intelRealsense\python\examples\box_dimensioner_multicam\realsense_device_manager.py", line 289, in get_depth_to_color_extrinsics
    frameset[rs.stream.infrared].get_profile())
KeyError: stream.infrared

honghande, 2019-01-09:
RealSense Viewer Can run

[image]

RealSenseCustomerSupport, 2019-01-23:
Hi @honghande,

This should be similar issue with #3032. Did you modified it based on left camera instead of color camera?
Suggest to close this and track on #3032.

closed 2024-03-29 by MartyG-RealSense