realsenseai/librealsense #3029: how to save a video recorded by RealSense

MaryBD, 2019-01-09:
Hi. Would you advice me how I can save a video recorded by realsense. I am using python. thanks.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          |

HippoEug, 2019-01-09:
You could try using `config = rs.config()` with `config.enable_record_to_file('test.bag')`

dorodnic, 2019-01-09:
Hi @MaryBD 
Doing `enable_record_to_file` as @HippoEug suggested will record the stream into SDK format (`.bag`) that can be viewed using the RealSense Viewer (or consumed from python).
Do you mean how to save video into some other specific format?

MaryBD, 2019-01-10:
Done! Thanks.
(Yes I meant .bag file)

MaryBD, 2019-01-10:
I am following distance_to_object tutorial but I cannot use the saved .bag file. I get this error: RuntimeError: Failed to resolve request. Request to enable_device_from_file("object_detection.bag") was invalid, Reason: Failed to create ros reader: Bag unindexed

I used ' cfg.enable_device_from_file("../object_detection.bag") ' for recording the video. Do I need to give it format or do that in the other way? thanks.

MaryBD, 2019-01-10:
I tried to close it from main and now the .bag file works. Thank you. But I still have an issue, the line: depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
works properly for the offline stream (.bag) and got error when I use camera online: NameError: name 'profile' is not defined

MaryBD, 2019-01-10:
I opened a new issue for "profile" error so I am going to close this issue.

RealSenseCustomerSupport, 2019-01-18:
@MaryBD Could you please paste more code snippet about how you define the profile?

pncanas333, 2019-01-31:
Hi MaryDB,
I´m having the same problem.. can you please share how you closed it from main?

MaryBD, 2019-02-04:
Here is a code for recording the stream:

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_record_to_file('object_detection.bag')

# Start streaming
pipeline.start(config)

e1 = cv2.getTickCount()

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
        e2 = cv2.getTickCount()
        t = (e2 - e1) / cv2.getTickFrequency()
        if t>30: # change it to record what length of video you are interested in
            print("Done!")
            break

finally:

    # Stop streaming
    pipeline.stop()

MaryBD, 2019-02-04:
Here is the code, I also updated the forum.

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_record_to_file('object_detection.bag')

# Start streaming
pipeline.start(config)

e1 = cv2.getTickCount()

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to
8-bit per pixel first)
        depth_colormap =
cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03),
cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
        e2 = cv2.getTickCount()
        t = (e2 - e1) / cv2.getTickFrequency()
        if t>30:
            print("Done!")
            break

finally:

    # Stop streaming
    pipeline.stop()

arturoribessanz, 2021-05-26:
Still no answer on this one? I am experiencing the same in C# but still have not found a workaround like here is done in python.

closed 2024-03-29 by MartyG-RealSense