realsenseai/librealsense #3030: How to convert the recorded frame to RGB or BGR

MaryBD, 2019-01-09:
I am streaming a video using realsense D435. I receive data frame by frame would you advice me about the content of each frame and how I can convert them to RGB and BGR to be enable of using them in opencv. I use python 3.6 on win 10. Thanks.

dorodnic, 2019-01-09:
When configuring live camera you can use `RS2_FORMAT_BGR8`. When working with a recording, you can do `destRGB = cv2.cvtColor(srcBGR, cv2.COLOR_BGR2RGB)` to convert using OpenCV

MaryBD, 2019-01-10:
Do you mean when using config.enable_stream(...)? Would you write a complete line of code for using RS2_FORMAT_BGR8 thanks.

dorodnic, 2019-01-10:
Hi @MaryBD 
Yes. Shortest way to achieve this is:
```cpp`
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR8);
rs2::pipeline p;
p.start(cfg);
```
Please note that the SDK examples will not know to display such frame, but you could modify [save-to-disk](https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk) to see the effect. 
You could also be more explicit in your call to configure resolution and FPS:
```cpp
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);
```

MaryBD, 2019-01-10:
Done! Thanks

closed 2019-01-10 by MaryBD

MaryBD, 2019-01-10:
Done! Thank you.