realsenseai/librealsense #3031: D435 Pointcloud

SunshineDou, 2019-01-09:
I use D435 to get aligned depth and rgb image. the depth unit is set to 0.00025. I use the "Kintinuous"to generate pointcloud, but the result is not very good, even when the object was about 2 meters away from the camera, but the pointcloud drift noticeably. What's the matter? I think it's the depth image's problem, as I want to reconstruct a room about 6mx10m, So how to set the camera to get a more accurate depth image?
[image]

Thank you very much

dorodnic, 2019-01-09:
First, D435 depth is not perfect. Wider field of view means less pixel density compared to the D415. 
That said, there are ways to improve it. The topic has been discussed in various issues and on RealSense Communities forum if you want the full story, but we also published [Best Known Methods for Tuning Intel RealSense Depth Cameras D415 and D435](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) whitepaper if you want the summary.

Mikeraka, 2019-01-09:
@dorodnic Thank you very much for your replying. I will have a look at the whitepaper first. Thank you

closed 2019-02-20 by SunshineDou