realsenseai/librealsense #3032: box_dimensioner_multicam__demo.py，RuntimeError: Couldn't resolve requests？

honghande, 2019-01-09:
My binocular camera only has two infrared cameras, no color cameras, but one of the infrared cameras is with color. How can I modify such hardware in the program?
My camera information as shown above: I want to open the infrared color camera, but the device seems to open all the transmitters, how to modify it, I modify the following:

Enable the streams from all the intel realsense devices
rs_config = rs.config()
rs_config.enable_stream(rs.stream.depth, resolution_width, resolution_height, rs.format.z16, frame_rate)
rs_config.enable_stream(rs.stream.infrared, 1, resolution_width, resolution_height, rs.format.rgb8, frame_rate)

rs_config.enable_stream(rs.stream.infrared, resolution_width, resolution_height, rs.format.y8, frame_rate)
Errors were reported as follows:
1 devices have been found
Traceback (most recent call last):
File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 158, in 
run_demo()
File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 48, in run_demo
device_manager.enable_all_devices()
File "E:\intelRealsense\python\examples\box_dimensioner_multicam\realsense_device_manager.py", line 168, in enable_all_devices
self.enable_device(serial, enable_ir_emitter)
File "E:\intelRealsense\python\examples\box_dimensioner_multicam\realsense_device_manager.py", line 153, in enable_device
pipeline_profile = pipeline.start(self._config)
RuntimeError: Couldn't resolve requests

dorodnic, 2019-01-09:
You are using `rs_config.enable_stream(rs.stream.infrared, 1, resolution_width, resolution_height, rs.format.rgb8, frame_rate)` instead of `rs_config.enable_stream(rs.stream.infrared, resolution_width, resolution_height, rs.format.y8, frame_rate)`, right? They will not work together. Did you try omitting the stream index (1) in `enable_stream` or changing it to zero?

honghande, 2019-01-10:
Hi，@dorodnic
Now I only have two infrared cameras, one with color infrared. How can I modify the program to achieve color measurement?

RealSenseCustomerSupport, 2019-01-21:
Hi @honghande,

Can you take a picture of your camera device? Are you a NDA customer?
The depth map is aligned on left camera. No extrinsic is required.

RealSenseCustomerSupport, 2019-01-22:
Hi @honghande,

You still can use the stereo camera to have the measurement (refer to "rs-measure" example). However, D405 depth range is very limited.
By the way, what's your company name? With NDA, we can further discuss the support of D405 project.

honghande, 2019-01-22:
Hi，RealSenseCustom，O-film

RealSenseCustomerSupport, 2019-01-22:
Hi @honghande,

As I know, this should be a customized D405 camera built by your company, right? Sine you are a NDA customer, suggest to file the issues on the Zendesk.