realsenseai/librealsense #3033: align color2depth in python

Mikeraka, 2019-01-09:
Hello, I want to align color to depth, my code is as follows:

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)

profile = pipeline.start(config)

intr = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
print("ppx is: ",intr.ppx)#The cameras principle point are descried by ppx and ppy, thus correspond to cx and cy respectively which is usally used in literature
print("ppy is: ",intr.ppy)
print("fx is: ",intr.fx)
print("fy is: ",intr.fy)

depth_sensor = profile.get_device().first_depth_sensor()

preset_range = depth_sensor.get_option_range(rs.option.visual_preset)
print('preset range:'+str(preset_range))
for i in range(int(preset_range.max)):
        visualpreset=depth_sensor.get_option_value_description(rs.option.visual_preset,i)
if visualpreset == "custom":
        depth_sensor.set_option(rs.option.visual_preset, i)

if depth_sensor.supports(rs.option.depth_units):
	depth_sensor.set_option(rs.option.depth_units,0.00025)

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

align_to = rs.stream.depth
align = rs.align(align_to)

i=1
j=1
try:
    while True:
        frames = pipeline.wait_for_frames()

        aligned_frames = align.process(frames)

        aligned_color_frame = aligned_frames.get_color_frame() 
        depth_frame = aligned_frames.get_depth_frame()
        
        if not aligned_color_frame or not depth_frame:
            continue
        
        frame = decimation.process(depth_frame) 
        frame = depth_to_disparity.process(depth_frame)
        frame = spatial.process(frame)
        frame = temporal.process(frame)
        filtered_depth_frame = disparity_to_depth.process(frame)

        depth_image = np.asanyarray(filtered_depth_frame.get_data())
        color_image = np.asanyarray(aligned_color_frame.get_data())
        
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
     
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.imwrite("//home//cindy//align2//TestAlign//depth//"+str(i)+".png",depth_image)
        cv2.imwrite("//home//cindy//align2//TestAlign//rgb//"+str(i)+".png",color_image) 
        i = i+1
        j = j+1
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()

but the result was as follows:
[image]

[image]

what's the matter that the rgb image was not aligned and has many black blocks in it?
Thank you very much

Mikeraka, 2019-01-09:
Why there are black pixels in color image? how to eliminate it? Thank you very much

lramati, 2019-01-09:
This is a side effect of aligning to depth. Align reprojects the color
frame so it looks like what you'd capture if the color camera was
physically located where the depth one was. However, there are sometimes
blind spots where the depth camera can see something that the color image
can't. In this case, the align function inserts black pixels into these
holes.
If it's important that the color image not have holes, you can align depth
to color instead, but keep in mind that then the aligned depth frame will
have holes, meaning there will be color pixels for which you don't have
depth data.

Mikeraka, 2019-01-09:
@lramati OK I have got it, Thank you very much for your explanation

closed 2019-01-10 by Mikeraka