realsenseai/librealsense #3036: D435 Change Resolution using Matlab error

RobotinoUser, 2019-01-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win10  |                                     | 
| Platform                                 | PC |
| Language                            |  Matlab                          | 
| Segment			|  Robot                      | 

### Issue Description

I have a problem changing the resolution of my D435 Depth Camera with the enable_stream() function. I found some other similiar topics, but i didnt find a solution.  

I tried the following code:

_**cfg = realsense.config();
cfg.enable_stream(realsense.stream.color,-1, 1280, 720, realsense.format.rgb8, 0);**_

Unfortunatly i alwys get the following errors:

_**Error using realsense.stream
Cannot find a member of the 'realsense.stream' enumeration class that corresponds to each element of the given input argument.**_

_**Error in realsense.config/enable_stream (line 86)
            args = num2cell(int64([varargin{:}]));**_

_**Error in streaming_final (line 9)
    cfg.enable_stream(realsense.stream.color,-1, 1280, 720, realsense.format.bgr8, 0);**_

Does anyone have an idea how to solve this error?

Thanks for your help!

lramati, 2019-01-16:
Please update your install. This issue was first reported in Issue #2487 and fixed in PR #2492

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

RobotinoUser, 2019-01-17:
> Please update your install. This issue was first reported in Issue #2487 and fixed in PR #2492

Thank you, it worked. And sorry for the double thread, i didnt see the similarities..

So the issue is solved!

closed 2019-01-17 by RobotinoUser

hshorty, 2020-07-06:
I have a problem getting the 'vertices' of my D435 Depth Camera after aligning the depth frame to colour frame. The error I get is as follow. (Sometimes the code works fine). I do change the resolution of my camera. As I noted 'pnts.get_vertices()' works fine when I don't align the frames.

**Error using librealsense_mex
null pointer passed for argument "frame"

Error in realsense.points/get_vertices (line 13)
            vertices = realsense.librealsense_mex('rs2::points', 'get_vertices', this.objectHandle);

Error in RealSencepredic_track_redObject_V3_30_06_2020 (line 140)
            vertices = pnts.get_vertices();  % HS [x y z]**

Does anyone have an idea on solving this issue?