realsenseai/librealsense #3037: how can i load Json file or enable post processing (Hole filling) by writing in my code ?

Medo91code, 2019-01-09:
Hi 
I use D435 , c++ as a programing language.
how can i enable post processing (Hole filling) in my code (manually by writing )?
or how  can i save the configuration of post processing (Hole filling) from RealSense Viewer as JSON file and after that load this file in the code ?

Thank you

dorodnic, 2019-01-09:
[post-processing](https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing)
We don't have an example for loading JSON, but you could do it like this:
```cpp
rs2::pipeline p;
auto prof = p.start();
rs2::device dev = prof.get_device();
auto advanced = dev.as<advanced_mode>();
advanced.load_json(str);
```
You could also apply one of the standard presets by setting `VISUAL_PRESET` option on the depth sensor.

Medo91code, 2019-01-09:
i tried what you wrote : 

rs2::pipeline p;
auto prof = p.start();
rs2::device dev = prof.get_device();
auto advanced = dev.as<advanced_mode>();
advanced.load_json(str);

but the compiler can not identify "advanced_mode" in (auto advanced = dev.as<advanced_mode>();).

HippoEug, 2019-01-10:
For hole filling, look up:

```
rs2::hole_filling_filter holeFilter;
...
...
data = pipe->wait_for_frames().apply_filter(holeFilter);
```

dorodnic, 2019-01-10:
@Medo91code 
You might need `using namespace rs2;`

Medo91code, 2019-01-13:
@dorodnic  
> You might need using namespace rs2;
i used it , but nothing happens ( dev.as<advanced_mode>(); still doesn't work).

@HippoEug 

> data = pipe->wait_for_frames().apply_filter(holeFilter);

it works but how can i add to it the choice (fill from left)?

HippoEug, 2019-01-14:
That I do not know, maybe @dorodnic can give some insights 👍

dorodnic, 2019-01-14:
For advanced mode, you need `namespace rs400` - [rs_advanced_mode.hpp:10](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rs_advanced_mode.hpp#L10)
Please review the linked post-processing example to learn how to configure filter settings.

closed 2019-04-07 by dorodnic