realsenseai/librealsense #3040: Install the Camera on Xilinx Ultra96 FPGA

abdelrahmanhussein, 2019-01-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435 | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |  Petalinux  | 
| Kernel Version (Linux Only)    |  4.15.0-42-generic |                                         | 
| Platform                                 | Xilinx Ultra96 FPGA  |
| SDK Version                            | 2.17.0                         | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue Description

Hello,

I am not sure if my question is going to be relevant to your scope, but let's try anyway!

I am working on connecting my D435 Camera to Xilinx Ultra 96 FPGA, which has a petalinux OS running on it already. If someone is familiar with FPGAs, how can I get to run any example on the board (rs-depth for instance)?
By that question I mean the following:

1. what files/libraries/packages do I have to install on the petalinux and how? I am just confused whether I should install the SDK and/or the librealsense?

2. I need some general guidelines about where to start. I know this is not an FPGA supporting forum, but I just hope there would be someone who has/had this kind of experience here.

Thanks.

MartyG-RealSense, 2019-01-10:
Is your board the one shown in the image below, please?

[image]

If so, you could likely get the Librealsense SDK to run on it by using a 'bypass' instruction that enables the SDK to be used with any kind of Linux.  Instructions for doing so are in the link below.

https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666

dorodnic, 2019-01-10:
Hi @abdelrahmanhussein 
I have zero experience with FPGAs, but assuming it runs a flavor of linux, most of librealsense (or RealSense SDK 2.0 as it is now called) will work. 
There are several options and constraints:
1. If you have libusb installed on the system, both LibUVC and V4L2 backends should compile (the V4L2 is default, you can switch to LibUVC using `-DFORCE_LIBUVC=true` as Marty suggested
2. If you don't have libusb, you can use `-DUSE_SYSTEM_LIBUSB=false` to make SDK build process pull libusb from the official git repo, build and install it.
> [Which Linux kernel is PetaLinux Tools using?](http://www.zachpfeffer.com/single-post/2018/05/10/Which-Linux-kernel-is-PetaLinux-Tools-using)
3. If you have Linux kernel 4.4+ the camera should be functional with the default V4L2 backend
4. If you have Linux kernel < 4.4, you can `-DFORCE_LIBUVC=true` and everything should work
5. If you have Linux kernel 4.16+ the camera should be **fully** functional with the default V4L2 backend
6. If you have Linux kernel >= 4.4 && <= 4.16, some camera functionality such as hardware timestamp will require kernel patching. We distribute [large selection](https://github.com/IntelRealSense/librealsense/tree/master/scripts) of patches adapted for different kernel flavors. The provided scripts, however, are likely not to work. Unless your kernel is monolithic, you should be able to build and replace the relevant modules manually (using `modprobe`)
7. The SDK does require GCC 5 for compilation, but otherwise is self-contained with the exception of libusb (which is LGPL)
8. SDK tools and wrappers are likely to depend on more different 3rd-parties, but you only need `realsense2.so` to work with the camera

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

MartyG-RealSense, 2019-01-20:
I'm not aware of special instructions being needed to use Librealsense with the GCC compiler specifically.  Instead, the documentation provides the following instructions for installing an Arch-based distribution.

1.  You need to install the **base-devel** package group.

https://www.archlinux.org/groups/x86_64/base-devel/

2.  You also need to install the matching linux-headers as well (i.e: linux-lts-headers for the linux-lts kernel).

3.  Navigate to the scripts folder 

cd ./scripts/

4.  Then run the following script to patch the uvc module: 

./patch-arch.sh

Regarding your question about build location ... I may be wrong about this, but I would think that the location of the build folder would be relative to where on your computer you installed the Librealsense package.  For example, after Librealsense is installed, the documentation asks you to create a build directory and navigate to it with the command **mkdir build && cd build**

So if you installed Librealsense in the file system of Petalinux, I would therefore expect your created build directory's location to also be in the Petalinux file system.

As I said, I can't be certain about that (I'm not a Linux user, I'm a Windows guy) - Dorodnic or one of the other Intel guys can provide confirmation (or correction).  :)

abdelrahmanhussein, 2019-01-20:
@MartyG-RealSense ,

I totally appreciate that. I will take a look into that as soon as I am at my office.

Regarding the building and installation directory, well, you are not completely wrong. I am asking that because after building and installing the library, I have run the `Which` command to see the directory of the package. All of the librealsense components and examples are found in usr/local/ directory, in spite of the fact that the main source files exist in my home directory.

However, the examples are also built in the build folder.

Regarding using gcc, well there are other aarch64-linux-.. compilers, and gcc is one of them.

MartyG-RealSense, 2019-01-20:
@RealSense-Customer-Engineering said in a past case that "You can define the flag "CMAKE_INSTALL_PREFIX" to assign the installation path you want".

You can find guidance on using this flag by going to the link below and scrolling halfway down th epage to the section headed 'Build librealsense'.

https://gist.github.com/svenevs/f00ed3898d2af6248921b63254aa8cc1

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

Hi @abdelrahmanhussein,

Still any trouble for the installation? If no more support for this topic, will close this.

closed 2019-03-05 by RealSenseCustomerSupport