realsenseai/librealsense #3041: What is "profile" in example: distance_to_object.ipynb

MaryBD, 2019-01-10:
I am following distance_to_object.ipynb and I got two issues. First the term profile in line ''depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()'' is shown as unresolved reference. Second if I use plt.imshow nothing appear and rather I need to use cv2.imshow. Please advice me. Thanks.

dorodnic, 2019-01-10:
Hi @MaryBD 
If you [try this link](https://hub.mybinder.org/user/intelrealsense-librealsense-98fv84h3/notebooks/notebooks/distance_to_object.ipynb) you can run the example inside your browser in a sandbox python environment. I'm not familiar with the unresolved reference error, could you please share the exact message? Perhaps you have older SDK version installed somewhere on your system?

MaryBD, 2019-01-10:
The link doesn't work. I am using SDK 2.0 and pyrealsens2 on python 3.6. This is the error:

Traceback (most recent call last):
  File "C:/Users/Maryam/PycharmProjects/TestOpenCV/ObjectDetectionRealSense.py", line 148, in <module>
    depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
NameError: name 'profile' is not defined

This line of code works well with offline data but when I use camera online it returns error.

dorodnic, 2019-01-10:
Sorry, the right link is:
https://mybinder.org/v2/gh/IntelRealSense/librealsense/jupyter?filepath=notebooks/distance_to_object.ipynb 

`profile` is being defined in the beginning of the script:
```py
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file("../object_detection.bag")
profile = pipe.start(cfg)
```
It represents the current hardware configuration captured on start.

MaryBD, 2019-01-10:
It works properly now. Thank you.

closed 2019-01-10 by MaryBD