realsenseai/librealsense #3042: Color stream cannot work at 1920x1080 resolution, showing "Incomplete video frame detected"

oiramario, 2019-01-10:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415                                                     | 
| Firmware Version                   |  05.11.01.00                                           | 
| Operating System & Version |  Ubuntu 18.04                                       | 
| Kernel Version (Linux Only)    |  4.4.143                                                 | 
| Platform                                 | nanopi m4 (rockchip RK3399)               |
| SDK Version                            | 2.17.1                                                   | 
| Language                               |  c++                                                      | 
| Segment			         |  AR                                                        | 

RK3399 board: 
4G ram
16G eMMC
CPU: big.LITTLE，Dual-Core Cortex-A72(up to 2.0GHz) + Quad-Core Cortex-A53(up to 1.5GHz)
GPU: Mali-T864 GPU，supports OpenGL ES1.1/2.0/3.0/3.1, OpenVG1.1, OpenCL, DX11, and AFBC
USB 3.0: four USB 3.0 Type-A ports
Power supply: DC 5V/3A

### Issue Description
rs-enumerate-devices
[image]

I plug D415 into USB3.0. Using GLESv2 + GBM + DRM to show color stream with Zero-Copy.
For the purpose of reducing the transmission bandwidth, I set the format RS2_FORMAT_YUYV to display the color stream.
When set resolution to 1280 x 720 it can running fast with 30fps.
However, when the resolution is set to 1920 x 1080, the log shows a lot of Incomplete video frame detected.
I have reviewed the description of such a problem in the forum, pointing out that the problem is due to insufficient usb capacity, or the CPU load is too high.
However, the official recommendation of intel to run the D400 device is AAEON UP Board (see link below), its ability is equivalent to RK3399, so I don't think it is the reason for hardware capabilities.
[https://click.intel.com/intelr-realsensetm-depth-camera-d415-aaeonr-up-board.html](https://click.intel.com/intelr-realsensetm-depth-camera-d415-aaeonr-up-board.html)

Also, what is the native format of the D415 color stream? Is it YUYV or UYVY? If it is because of the lack of computing power due to format conversion, or the problem of USB transmission bandwidth, I can directly use the native format and then quickly convert through GPU.
Please give your suggestion.

Below is the relevant logo information

10/01 12:52:49,118 INFO [548547798704] (backend-hid.cpp:1178) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 10/01 12:52:49,118 DEBUG [548547798704] (ds5-factory.cpp:568) try_fetch_usb_device(...) failed.
 10/01 12:52:49,123 INFO [548547798704] (backend-v4l2.cpp:1468) v4l_uvc_meta_device
 10/01 12:52:49,127 INFO [548547798704] (backend-v4l2.cpp:1468) v4l_uvc_meta_device
 10/01 12:52:49,136 INFO [548547798704] (backend-v4l2.cpp:1468) v4l_uvc_meta_device
 10/01 12:52:49,141 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format Depth data 16-bit (Z16)
 10/01 12:52:49,147 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format 8-bit Greyscale
 10/01 12:52:49,148 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format UYVY 4:2:2
 10/01 12:52:49,148 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format 8-bit Greyscale
 10/01 12:52:49,154 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format Greyscale 8 L/R (Y8I)
 10/01 12:52:49,160 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format Greyscale 12 L/R (Y12I)
 10/01 12:52:49,172 DEBUG [548547798704] (backend-v4l2.cpp:1254) Recognized pixel-format YUYV 4:2:2
 10/01 12:52:49,179 WARNING [548547798704] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 10/01 12:52:49,180 WARNING [548547798704] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
 10/01 12:52:49,206 INFO [548547798704] (backend-v4l2.cpp:1460) Trying to configure fourcc YUYV
 10/01 12:52:49,206 DEBUG [548547798704] (backend-v4l2.cpp:1587) Metadata node was successfully configured to D4XX format
 10/01 12:52:49,237 INFO [548170531248] (backend-v4l2.cpp:996) Empty video frame arrived
 10/01 12:52:49,239 INFO [548170531248] (backend-v4l2.cpp:996) Empty video frame arrived
 10/01 12:52:49,239 INFO [548170531248] (backend-v4l2.cpp:996) Empty video frame arrived
 10/01 12:52:49,239 INFO [548170531248] (backend-v4l2.cpp:996) Empty video frame arrived
 10/01 12:52:49,242 INFO [548547798704] (backend-v4l2.cpp:1473) ~v4l_uvc_meta_device
 10/01 12:52:49,242 INFO [548547798704] (backend-v4l2.cpp:1473) ~v4l_uvc_meta_device
 10/01 12:52:49,247 INFO [548547798704] (backend-v4l2.cpp:1473) ~v4l_uvc_meta_device
 10/01 12:52:49,383 WARNING [548170531248] (backend-v4l2.cpp:1007) FD_ISSET returned false - video node is not signalled (md only)
 10/01 12:52:49,386 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,416 WARNING [548170531248] (backend-v4l2.cpp:1007) FD_ISSET returned false - video node is not signalled (md only)
 10/01 12:52:49,419 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 181000 out of 4147455 bytes (4%)
 10/01 12:52:49,450 WARNING [548170531248] (backend-v4l2.cpp:1007) FD_ISSET returned false - video node is not signalled (md only)
 10/01 12:52:49,453 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 181000 out of 4147455 bytes (4%)
 10/01 12:52:49,490 WARNING [548170531248] (backend-v4l2.cpp:1007) FD_ISSET returned false - video node is not signalled (md only)
 10/01 12:52:49,491 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 1098504 out of 4147455 bytes (26%)
 10/01 12:52:49,520 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,554 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,587 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,622 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,655 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 213768 out of 4147455 bytes (5%)
 10/01 12:52:49,687 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 181000 out of 4147455 bytes (4%)
 10/01 12:52:49,719 INFO [548170531248] (types.h:936) Incomplete video frame detected!
Size 181000 out of 4147455 bytes (4%)
 10/01 12:52:49,752 INFO [548170531248] (types.h:936) Incomplete video frame detected!

dorodnic, 2019-01-10:
Hi @oiramario 

For the RGB sensor, the native format is YUYV (fourcc 'YUY2'). 

The D415 also offers lower quality RGB stream from the left infrared camera. This stream is available at up to 720p and suffers from projector interference. On the bright side, it is pixel-aligned with depth and projector pattern can be mostly removed with a special preset. For this stream the format is UYVY. 

If I understand the problem correctly, you are getting 0 complete frames at 1080p from V4L2 at all, right? 
Is this only when depth streaming as well, or also when streaming color alone?

I do believe this capability was functional on the Up-Board. If you'd like, I can double-check, but don't have one next to me right now.

dorodnic, 2019-01-10:
Something *I think* might be relevant - [LKML Asynchronous UVC](https://lkml.org/lkml/2018/1/3/698)
To rule out librealsense, you could test the system using something like [yavta](https://github.com/fastr/yavta)
In general, the `Incomplete video frame detected!` warning originates in the `uvc_video` Kernel Module (it is a flag on the buffer object if I recall correctly). However, if we forgot a thread doing `while(true);` somewhere in librealsense it would certainly be a problem :)
Otherwise, librealsense should add very little overhead when configured as you described. 
@ev-mp

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

oiramario, 2019-01-18:
Not yet. I'll try the patch next week.

RealSenseCustomerSupport, 2019-01-21:
@Oiramario Good. Looking forward to your update!

RealSenseCustomerSupport, 2019-01-23:
@Oiramario  Please try to add below two patches.

**1. modify_urb_number_and_uvc_packet.patch**
--- a/drivers/media/usb/uvc/uvcvideo.h
+++ b/drivers/media/usb/uvc/uvcvideo.h
@@ -172,9 +172,9 @@
 #define DRIVER_VERSION  "1.1.1"

 /* Number of isochronous URBs. */
-#define UVC_URBS  5
+#define UVC_URBS  32
 /* Maximum number of packets per URB. */
-#define UVC_MAX_PACKETS  32
+#define UVC_MAX_PACKETS  16
 /* Maximum status buffer size in bytes of interrupt URB. */
 #define UVC_MAX_STATUS_SIZE    32
 #define UVC_CTRL_CONTROL_TIMEOUT    300

**2. rk3399-irqchip-xhci-irq-142-to-all-cpu.patch**

--- a/drivers/irqchip/irq-gic-v3.c
+++ b/drivers/irqchip/irq-gic-v3.c
@@ -421,6 +421,12 @@ static void __init gic_dist_init(void)
  affinity = gic_mpidr_to_affinity(cpu_logical_map(smp_processor_id()));
  for (i = 32; i < gic_data.irq_nr; i++)
      gic_write_irouter(affinity, base + GICD_IROUTER + i * 8);
+
+    /* irq 142 routed to all cpus  */
+    affinity = affinity | 0x80000000;
+    gic_write_irouter(affinity, base + GICD_IROUTER + 142 * 8);
+    /* Decrease irq 142 priority value from default 0x0a to 0x09 */
+    writel_relaxed(0xa090a0a0, base + GICD_IPRIORITYR + rounddown(142, 4));
 }

 static int gic_populate_rdist(void)
@@ -658,7 +664,8 @@ static int gic_set_affinity(struct irq_data *d, const struct cpumask *mask_val,
  reg = gic_dist_base(d) + GICD_IROUTER + (gic_irq(d) * 8);
  val = gic_mpidr_to_affinity(cpu_logical_map(cpu));

-    gic_write_irouter(val, reg);
+    if (gic_irq(d) != 142)
+  gic_write_irouter(val, reg);

  /*
  * If the interrupt was enabled, enabled it again. Otherwise,

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:04 PM

@oiramario Any good news? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@oiramario Can the patch work for you? Looking forward to your result. Thanks!

RealSenseCustomerSupport, 2019-02-15:
----------------------------------------------

@oiramario Does it works? Please update. Thanks!

oiramario, 2019-02-15:
@RealSenseCustomerSupport I just came back from vacation and I will check this out next week.

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

@oiramario Any update? Thanks!

oiramario, 2019-02-22:
Sorry, the confirmation of the patch is delayed because of some other work.
There is a problem with the patch content you posted, and it cannot be used directly after copying it. Referring to the code, I re-created the patch and successfully patched it into the kernel. However, I found that in the kernel v4.4.143 used by rockchip, the macro UVC_MAX_STATUS_SIZE is 16, and 32 of the content you sent, I am not very clear about this part, I hope you can help confirm it.
Next week I will confirm the modified kernel on the board of rockchip 3399, thank you.

[modify_urb_number_and_uvc_packet.txt](https://github.com/IntelRealSense/librealsense/files/2893859/modify_urb_number_and_uvc_packet.txt)
[rk3399-irqchip-xhci-irq-142-to-all-cpu.txt](https://github.com/IntelRealSense/librealsense/files/2893860/rk3399-irqchip-xhci-irq-142-to-all-cpu.txt)

oiramario, 2019-02-22:
For your reference:
https://github.com/rockchip-linux/kernel/blob/release-4.4/drivers/media/usb/uvc/uvcvideo.h

[image]

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

@oiramario The difference I think is due to different kernel version. You can have a try with your setting of UVC_MAX_STATUS_SIZE

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@oiramario How's it going? Any update? Thanks!

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@oiramario Any good news?

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@oiramario Now unroot android solution is included in 2.19 librealsense. No root needed and no patch needed. Please have a try to see if this can fix your issue. Thanks!
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

landersson, 2019-03-08:
@oiramario , sorry to hijack this thread... I'm trying to do what you seem to have succeeded in doing, i.e getting a RealSense device to work on a NanoPI M4. Please let me know if you would be happy to outline the basic steps that are required... I have successfully build the realsense libs/tools and I can grab images from some of the /dev/video* devices created, but when trying the rs-enumerate-devices script I'm just getting "No device detected. Is it plugged in?"... I guess I have to patch and rebuild a custom kernel?

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

@landersson  What's OS you getting the issue, android or linux?

landersson, 2019-03-08:
I'm using the the standard Ubuntu 18.04 based "FriendlyDesktop" image for NanoPI M4. Default kernel 4.4.143. Basically the same setup as oiramario initially reported.

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@oiramario Sorry, my mistake. I've thought the issue was found on Android. Will check this issue on Ubuntu and update you later.

oiramario, 2019-03-13:
> @oiramario , sorry to hijack this thread... I'm trying to do what you seem to have succeeded in doing, i.e getting a RealSense device to work on a NanoPI M4. Please let me know if you would be happy [...]

Follow these steps:
```
1. apt-get install git libssl-dev libusb-1.0-0-dev pkg-config
2. git clone --depth 1 https://github.com/IntelRealSense/librealsense.git
3. cd librealsense
4. sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
    sudo udevadm control --reload-rules && udevadm trigger
5. mkdir build && cd build
    cmake -DCMAKE_BUILD_TYPE=Release \
             -DBUILD_WITH_TM2=false -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_EXAMPLES=false
    make -j$(nproc) && make install 
6. rs-enumerate-devices
```

More detail:
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

oiramario, 2019-03-13:
> @oiramario Any good news?

Sorry, I have been stuck in some kernel-related issues recently, but I will take time to verify this issue as soon as possible.

landersson, 2019-03-15:
Thanks @oiramario, those are pretty much exactly the steps I took too, but I'm still getting "No device detected..." when running rs-enumerate-devices. Running as sudo doesn't help.

My camera is a 435i, so I guess that must be why we're getting different results. Apart from the camera model, our setups are nearly identical (except you have 4G RAM while I have 2).

landersson, 2019-03-15:
@oiramario , did you connect the depth camera via some kind of powered USB hub, or straight to the Nano PI M4? 

I'm thinking perhaps the NanoPI can't provide enough power via USB to run the D435i...

oiramario, 2019-03-20:
I didn't have any USB hub here,  just plug in D415 into NanoPi-M4.
Could you post some logs or shots?

landersson, 2019-03-21:
Ok, I tried with a powered USB3 hub, but that didn't change anything. I've attached an strace of rs-enumerate-devices, the output of "lsusb -vvv" and the end of dmesg... Let me know if there's anything else that can help you figure out why this doesn't work. Thanks.

[lsusb.txt](https://github.com/IntelRealSense/librealsense/files/2990511/lsusb.txt)
[dmesg-tail.txt](https://github.com/IntelRealSense/librealsense/files/2990512/dmesg-tail.txt)
[strace.txt](https://github.com/IntelRealSense/librealsense/files/2990514/strace.txt)

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@oiramario  Did you get chance to try again with the kernel patch?

@landersson  How many USB ports on your NanoPI? Did you try them all? Seen from the result of lsusb, the camera has been connected and recognized and rs-enumerate should work. Please double check if "sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ "  executed successfully.

landersson, 2019-03-21:
The Nano M4 have 4 USB3 ports.. I've tried them all as well as connecting the D435i using an external, powered USB3 hub.The udev script has been properly installed.

landersson, 2019-03-22:
Just took some time to look further into this... With logging enabled, I see this message:

22/03 03:19:33,133 WARNING [547723022352] (ds5-factory.cpp:617) DS5 group_devices is empty.

Looking at src/ds5/ds5-factory.cpp:591, it looks like my D435i is disabled because it's a HID device, but no HID sensors were detected. I assume the HID sensors (gyro and accelerometer) would be detected if with a patched kernel, so I guess I'll just have to go ahead and rebuild the kernel.

@oiramario, if you have any pointers in regards to rebuilding the kernel for the NanoPI M4, that would be great.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@landersson What's the FW version of your D435i? Could you please try to connect it to one PC to see if it's 5.10.13+ and if it can work normally? Please be noted that "D435i requires FW version 5.10.13+ and is currently supported on Linux and Windows 10 only. "

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@landersson If need HID related pacth, please refer to https://github.com/IntelRealSense/librealsense/tree/master/scripts

Please select the patch accordingly such as
https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-hid-ubuntu-bionic-master.patch

landersson, 2019-03-22:
realsense-viewer on my Ubuntu 18.04 PC says: Firmware Version:  05.10.13.00

landersson, 2019-03-22:
In order to end up with a kernel that is going to work well on the NanoPI M4 SOC, I assume I will have to use a custom kernel that have proper support for this board, such as:

https://github.com/friendlyarm/kernel-rockchip

This is based on Linux 4.4, so I guess I have to apply the realsense patches to this kernel manually.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@landersson Yes, you're right. You need apply the patch manually. Any update please let us know. Thanks!

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

@landersson  Any update? Thanks!

landersson, 2019-03-27:
Hi, I just applied the patches to the standard NanoPI M4 RockChip kernel (4.4.154)... I had to fix up a few of the changes manually. After booting with the patched kernel, rs-enumerate-devices finds my D435i and realsense-viewer works pretty well too.  I do get a lot of "Incomplete video frame detected" warnings if I try to use the depth stream in all but the lowest resolutions/frame rates. realsense-viewer also seem to be using up a fair bit of CPU even when no streams are enabled.

I haven't had time to investigate and reduce the number of incomplete frames... any suggestions are welcome. Color seem to be able to run as much higher resolutions without getting incomplete frames... Is the color stream being transferred as compressed video, while depth is uncompressed? Or is it more likely that CPU bottlenecks are causing incomplete depth frames?

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@landersson Glad to see the progress! What's the resolution and fps of depth when you see the incomplete video frame detected? Any other streaming at the same time when you get incomplete frames?

@oiramario Any progress from your side? Thanks!

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@landersson @oiramario Any update? Thanks!

landersson, 2019-04-09:
Regardless of whether the color stream is running or not, the depth stream in only running with out incomplete frames in the lowest resolution.

I have run out of time to work more on this for the moment, but we will probably get back to this at a later stage. As far as I'm concerned, please feel free to close this issue. I will open up a new issue related to NanoPI M4 depth stream issues later on if I'm unable to resolve the incomplete frame problems myself.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@oiramario If no other questions, will close the ticket accordingly. Thanks!

oiramario, 2019-04-29:
@RealSenseCustomerSupport 
I'm back!
I'll into this next week and feed back to you.

closed 2019-05-03 by RealSenseCustomerSupport