realsenseai/librealsense #3043: Documentation for alignment pipeline with the openni wrapper

yashmanian, 2019-01-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400} | 
| Operating System & Version | Ubuntu 16| 
| Platform                                 | PC |
| Language                            | C++| 

### Issue Description
I am looking to use the rs2::align pipeline through the openni wrapper for librealsense2. I haven't been able to find examples or documentation doing that at the moment. 
It would be great if someone could point me in the direction of more descriptive documentation.

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

wongfei, 2019-01-18:
@yashmanian 

check this one:
`m_device.setImageRegistrationMode(openni::IMAGE_REGISTRATION_DEPTH_TO_COLOR);`

https://github.com/OpenNI/OpenNI2/blob/master/Samples/SimpleViewer/Viewer.cpp#L268

RealSenseCustomerSupport, 2019-02-04:
----------------------------------------------

Hi yashmanian,

If there is nothing else, this one will be closed.

Thanks!

closed 2019-02-07 by yashmanian