realsenseai/librealsense #3044: Unable to open ros file after saving using pyrealsense2

jkenney9a, 2019-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435} | 
| Firmware Version                   | 05.10.13.00) | 
| Operating System & Version |   Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |4.15.0-43| 
| Platform                                 | PC|
| SDK Version                            | 2.17.1 | 
| Language                            |  {python 3.5.6 }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm having an issue with opening up a bag file with the realsense viewer after recording it in Python. I get this error: 

> Failed to load file ... test.bag. Reason: Io in rs2_context_add_device(ctx:0x1274510, file:/):
> Failed to create ros reader: Bag unindexed

This looks similar but not quite the same as issue [2326](https://github.com/IntelRealSense/librealsense/issues/2326) but I'm unable to resolve it.

The code is as follows:

```
import pyrealsense2 as rs
import time

pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
config.enable_record_to_file('test.bag')
pipeline.start(config)

try:
    start=time.time()
    while time.time() - start < 20:
        pipeline.wait_for_frames()
finally:
    pipeline.stop()
```
Oddly, I'm able to open the bag file if I run this code, and then re-run it and change 
`pipeline.start(config)` to `pipeline.start()`

HippoEug, 2019-01-11:
I am using C++ and experiencing similar issues. After a recording, I cannot open the file directly as I will get the same error. I've got to close and open my application (restart) in order to successfully open the bag file.

I think there is something wrong during the desconstruction of `enable_record_to_file` function/config. Some resources of some sort are not properly closed

jkenney9a, 2019-01-11:
Just wanted to give a brief update. I'm using Anaconda and Spyder for code development. This issue happens when I'm running the code in Spyder and executing in the IPython console. However, if I execute it from the shell (i.e. `python video_recording.py`), then there is no issue with opening the resulting file. This is a little annoying for code development, but at the end of the day this is how we plan on making use of the file so not as major an issue for us.

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

jkenney9a, 2019-01-18:
Well shouldn't running the command `pipeline.stop()` be sufficient to properly close and index the bag file? Sounds like this isn't python specific, as the same issue has been brought up with C++ [here](https://github.com/IntelRealSense/librealsense/issues/2835).

realsens01, 2019-02-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5_11_1 | 
| Operating System & Version |   Windows 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.17.0 and 2.17.1 |
| Language                            |  python                          | 
| Segment			|  others                          | 

I seem to be having an issue similar to @jkenney9a . I want to add that I'm not using an IDE, but I am using classes with pipeline and config variables that persist beyond the stop call. I did not notice this error when testing without classes. @dorodnic notes [elsewhere](https://github.com/IntelRealSense/librealsense/issues/1857#issuecomment-395870781) that _`Bag unindexed` error indicates that the recording was not properly closed_. I'm guessing that is what's happening, but I agree with @jkenney9a that the stop call should be sufficient to write the bag. I do not have access to the system right now, so I cannot test whether or not destroying the config and pipeline objects resolves the issue.

Also worth noting: ``rosbag reindex`` fixes corrupt bags, but it is an annoying intermediate step.

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi realsens01,

Thanks for the feedback.

I have not been able to reproduce the issue with python code that jkenney9a shared. Wonder if you can share out your sample and we will take a look at it.

Thanks!

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi realsens01,

I still can't reproduce this python one.
Wonder if you can share out snippet of your code to help debug this one?

Thanks!

realsens01, 2019-03-14:
I don't have constant access to the Realsense cameras, as we are using them to collect data for a study. This is a relatively low priority issue for me, because -- as expected -- destruction of the objects was a sufficient work-around. As @jkenney9a and I noted, a call to .stop did not write the bag properly, but a call to .stop plus the deliberate destruction of all Realsense objects does result in a working bag. I still consider this an issue, but not one that I can spend more time on.
 
Did you try @jkenney9a 's code in IPython? That seems like the simplest way to reproduce the error. IPython is an important part of the equation, as it is likely preventing the wrapper from cleaning up. As dorodnic noted, the issue is likely caused by wrappers that do not completely clean up Realsense objects. As @jkenney9a and I noted, this does not seem like expected or appropriate behavior. When we call .stop, we have the expectation that we've written a valid bag file. But the persistence of the Realsense objects (due to IPython in @jkenney9a 's case and persistent class variables in mine) seems to prevent this.

I can't test this right now, but the following code probably results in an improperly indexed bag:
```
import shutil, time, pyrealsense2 as rs2

class Pipeline:
    
    def __init__(self, filepath='test.bag'):
        self.filepath = filepath
        context       = rs2.context()
        self.device   = device = context.devices[0]
        serial_number = device.get_info(rs2.camera_info.serial_number)
        get_sensor    = lambda s: s.get_info(rs2.camera_info.name)
        self.sensors  = {get_sensor(s): s for s in device.sensors}        
        self.config   = config = rs2.config()
        self.pipeline = rs2.pipeline(context)
        config.enable_device(serial_number)
        config.enable_record_to_file('tmp.bag')
        
    def start(self): self.pipeline.start(self.config)
        
    def stop(self):
        self.pipeline.stop()
        
        # When the lines below are un-commented, the bag seems to write
        # properly. When they are commented, the bag produces an index error.
        
        #del self.pipeline
        #del self.config
        #del self.sensors
        #del self.device
        #self.pipeline = None
        #self.config = None
        #self.sensors = None
        #self.device = None
        
        shutil.move('tmp.bag', self.filepath)
        

if __name__ == '__main__':
    pipeline = Pipeline()
    pipeline.start()
    time.sleep(1)
    pipeline.stop()
```

realsens01, 2019-04-30:
The code I provided above didn't do quite what I expected. It returns this error: ``The process cannot access the file because it is being used by another process``. However, that still illustrates the problem (i.e., that the bag is not written when the pipeline is stopped). It's not a major issue for me anymore, but it was definitely frustrating when I didn't know that you have to destroy... everything?... before the bag is written. It's just undocumented, unexpected behavior.

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi realsens01,

Thanks for the updates and feedback. It is good to know that you resolved the issue.
Closing this one.

Thanks!

realsens01, 2019-05-22:
I didn't resolve the issue. I worked around it. This will likely still affect people that are not aware of it, and could potentially waste a lot of time -- as it did for me.

Did you reproduce the issue yet?

closed 2019-05-29 by RealSenseCustomerSupport

realsens01, 2019-05-29:
For the record, this just happened to me again. Ironic timing, given that @RealSenseCustomerSupport just closed the issue. I did not change my code, so I am not certain why the bag was "unindexed" this time. But the rosbag utility was able to re-index the file, so I did recover the data.

jamjahal, 2020-06-05:
I am also having trouble opening files on pyrealsense2.  I'm sharing files with someone else who has the camera, but bag files sent appear to be unindexed (no .bag extension).  I am unable to open these files, when I add the .bag extension, I get the following error:

Failed to create ros reader: Error reading from file: wanted 4 bytes, read 0 bytes

But then when I try to use[ sample files](https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md) provided by intel, in the [realsense colab API notebook](https://colab.research.google.com/drive/10YTLAf2i0R80-XX_6-1gTPtOMD0cRIx5) I see the same error!  

Required Info |  
-- | --
Camera Model | L515
Firmware Version | ?
Operating System & Version | Mac OSX 10.15.4 / Google Colab
Kernel Version (Linux Only) | N/A
Platform | PC / Colab Notebook
SDK Version | 2.34.0
Language | python 3.6.9

jkenney9a, 2020-06-20:
Sorry for my late reply. I never pursued this further; we primarily run things from the shell now, so it's no longer an issue. Was more of an annoyance than anything during code development.

lamuyang, 2023-08-18:
I try to use `rosbag reindex [bag file]` to fix and it work.

Install it first on linux with following command
`sudo apt install python3-rosbag`

MartyG-RealSense, 2023-08-18:
Thanks so much @lamuyang for sharing what worked for you!