realsenseai/librealsense #3045: Apparent misalignment after rs.align

ajprax, 2019-01-11:
**Camera Model**: D415
**Firmware Version**: 5.10.??? (can't check right now, but updated recently)
**Operating System & Version**: MacOS 10.14.2
**Platform**: PC
**SDK Version**: 2.2.17
**Language**: python
**Segment**: Robot

### Issue Description
I am noticing a misalignment between depth and color frames after using `rs.align`. My setup has objects fairly near the sensor (30cm) with custom disparity shift values in order to get depth readings at all. I'm wondering if being close to the sensor has any effect on the quality of the alignment. I also noticed that the alignment includes some scaling of the image, and the result is about the right size.

[image]
[image]

HippoEug, 2019-01-11:
Quick question, are you aligning color to depth or depth to color? Thanks!

ajprax, 2019-01-11:
Depth to color. I can try color to depth tomorrow.

ajprax, 2019-01-11:
Here are the results of aligning color to depth:

[image]
[image]

The black outline of the hands align perfectly, but you can still see a bit of fingertip protruding in the aligned color image.

ajprax, 2019-01-14:
I tried this, but there doesn't seem to be an Occlusion Removal option in my viewer. I'm using a D415 and Viewer 2.17.1 while the user from #2445 was using a D435 and Viewer 2.15.

[image]

ajprax, 2019-01-14:
I've also tried checking the alignment after rotating the hand 90 degrees. My thought is that occlusions issues should occur along the axis joining the sensors and that in this orientation both sensors should be able to see the tips equally well (and new occlusion issues should show up for the sides, this seems to agree with the location of the black "no data" regions in the depth image).

[image]
[image]

It's noteworthy that while the images still don't match, the discrepancy has changed orientations as well, suggesting it's related to the data itself in some way. My reading of the `align` code suggests it's fairly agnostic to the data being translated (excluding places where no depth data is present or where reprojected points conflict) so it may not be truly an alignment issue, but I still need to find a way to get the correct depth for a point of interest identified from the color image.

ajprax, 2019-01-14:
I was able to get the Occlusion Removal option to appear by switching to the color stream (just to the right of where the occlusion removal option shows) in 3d mode instead of the depth stream. Does turning this on change a setting in the camera similar to advanced mode options? Should I expect that images captured through the python API after I've enabled Occlusion Removal will include it? Some test images seem to show a pretty similar misalignment.

dorodnic, 2019-01-15:
Occlusion filter is part of the pointcloud post-processing. It only makes sense when mapping to a different viewport (for instance color). 

Could you please clarify if after applying occlusion filter in the Viewer the artifact was reduced / removed?

I assumed it can be related because it often explains duplication of texture to background, and having object close to the camera increases "dead-zones" that one of the sensors can't see. 

However, it is entirely possible there is a calibration misalignment, because camera calibration at production line is optimized given data at "regular" distances of 1-3 meters.

ajprax, 2019-01-16:
I was able to dramatically improve the agreement between the images by turning back on auto exposure for the depth sensor. I had set a fixed exposure for both sensors because I was having trouble with inconsistent rgb values in the color frame making it hard to localize a calibration marker.

I still see a little bit of the finger projected onto the table in the point cloud view, and changing the Occlusion Removal setting doesn't seem to have any effect. With the improvements I got today, this isn't my highest priority any more, so I'm going to close this. Thanks for the help.

closed 2019-01-16 by ajprax