realsenseai/librealsense #3047: Libusb fails at initializing (libusb_init)

Folyz, 2019-01-11:
Currently I've tried to build and run the realsense-viewer.exe for both the latest version (v2.17.2) as well as an older version v2.16.5. I've done this using Cmake and Visual Studio 2017 (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md). In the past I've succesfully compiled and run v2.11.0 with no issues.

Compilation goes fine, however as I start any program that uses the librealsense2.dll such as the realsense viewer it crashes. When compiling in debug mode it gives an access violation at /third-party/libusb/libusb/core.c line 2248:

usbi_backend.clock_gettime(USBI_CLOCK_REALTIME, &timestamp_origin);

I'm guessing the function clock_gettime to which is pointed to is not initialized or something else is wrong, however I couldn't trace back where this error could come from. I've been trying different kind of setups (all windows, visual studio) but I couldn't find the problem, any ideas?

dorodnic, 2019-01-12:
Hi @Folyz 
I also experienced this yesterday, and what worked for me was modifying [GIT_TAG 
 @ CMake/external_libusb.cmake:8](https://github.com/IntelRealSense/librealsense/blob/master/CMake/external_libusb.cmake#L8) from `master` to `v1.0.22` (and rebuilding all, by deleting the `build` folder and re-running CMake) I think something got broken in libusb. We will include the fix in upcoming release, but I hope for now this helps.

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

dorodnic, 2019-01-22:
Should be resolved now

closed 2019-01-22 by dorodnic