realsenseai/librealsense #3048: ROS rs_camera.launch causing "COMM_RXCORRUPT: Incorrect status packet!" on dynamixels !!

elpimous, 2019-01-11:
| Camera Model                       D435
| Firmware Version                   05.10.13.00
| Operating System & Version  ubuntu 16.04
| Platform                                   jetson TX2
| SDK Version                            LibRealSense v2.17.1, RealSense ROS v2.1.3
| Language                                python
| Segment			          robot : ROS Kinetic

ROBOT : USB 1, 2, 3
D435 : USB 0 (usb3)

### Issue Description

Hello,
My robot is running 2 dynamixels AX-12A dor head pan and tilt...

when launching only robot launch, no errors on terminal.

when launching : `roslaunch realsense2_camera rs_camera.launch`, 

i see warnings on my robot launch (and lost datas) : 
`
...
[ WARN] [1547244070.667192894]: COMM_RXCORRUPT: Incorrect status packet!
[ WARN] [1547244070.820253392]: COMM_RXCORRUPT: Incorrect status packet!
[ WARN] [1547244070.832341943]: COMM_RXCORRUPT: Incorrect status packet!
[ WARN] [1547244074.154946172]: COMM_RXCORRUPT: Incorrect status packet!
[ WARN] [1547244078.826000645]: COMM_RXTIMEOUT: There is no status packet!
...`

I don't know where could be the problem !! It seems that the D435 lib cause interferences with my robot system.

Any idea ??

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

elpimous, 2019-01-17:
??

RealSenseCustomerSupport, 2019-01-18:
Hi elpimous,

I on/off worked with ROS previous releases on an Intel NUC5i7RYH w/ Ubuntu 16.04, and am not aware of any side-effect that ROS causing to other things.
Not sure how is the internal bus structure of your robot? And how your application is implemented? Can you share more info?

Thanks!

RealSenseCustomerSupport, 2019-01-25:
Hi elpimous,

Is there any update from you?
If nothing else is needed, we are going to close this one.

Thanks!

elpimous, 2019-01-27:
Hello all !
SOLVED !

If you have frequent timeout errors, change l.155 of dxl_hal.c : 

`gfRcvWaitTime = (float)(gfByteTransTime*(float)NumRcvByte + 5.0f); `
to 
`gfRcvWaitTime = (float)(gfByteTransTime*(float)NumRcvByte + 34.0f);`

Hope it will help ! 
thanks to realsense team !

closed 2019-01-27 by elpimous

RealSenseCustomerSupport, 2019-01-28:
Hi elpimous,

Thanks for the update. Glad to know that it is solved.

Cheers!