realsenseai/librealsense #3050: How to place Device Setting file (.ini) for RealSense Devices on OpenNI application?

kana32kana, 2019-01-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5_10_13 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.17.0                          | 
| Language                            |  C++                          | 
| Segment			|  AR                          | 

I thought device setting files are should placed in "BIN_DIR/OpenNI2/Drivers/*.ini" in general OpenNI aplication.
I placed "rs2driver.dll" and "realsense2.dll" in that folder, it works well with D435, but don't know how I place setting file (.ini) for D435?

In wrapper souce of librealsence2, says rsdriver emulates "PS1080", but "PS1080.ini" in Drivers folder doesn't work for rsdriver.
I want to get human body skeltals on 5 - 6 meter away, so I want to edit parameters of device with trials and errors.

How I can place a device setting file for RealSense Device?

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

wongfei, 2019-01-18:
@kana32kana 
rs2driver doesn't support the config file, only runtime configuration via public OpenNI API

as easy workaround you can tune sensor options in RealSenseViewer and save json profile, then call rs2_load_json somewhere in OniStatus Rs2Device::startPipeline()

kana32kana, 2019-01-20:
Thank you wongfei.
So, You mean I should  call jason file in RsDevice.cpp? not in OpenNI Appli code tree?

wongfei, 2019-01-20:
`void rs2_load_json(rs2_device* dev, ...)`
dev handle available only inside RsDevice, have to patch rs2driver anyway (to expose handle or implement invoke command)

RealSenseCustomerSupport, 2019-01-25:
@kana32kana Any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:18 PM

@kana32kana Did you get it throught? Please update. Thanks!

closed 2019-02-15 by RealSenseCustomerSupport