realsenseai/librealsense #3060: can i build rs SDK 2.0 for android with ubuntu 18.04.1 LTS

cucrui, 2019-01-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When I build SDK 2.0 for Anddroid OS, I met a problem:

/aarch64-linux-android/bin/ld : skipping incompatible /home/ict/Downloads/librealsense-2.17.1/build/libusb_install/lib/libusb.a when searching for -lusb 

and finally I found that in instruction the required Linux host machine should with Ubuntu 16.04

dorodnic, 2019-01-14:
In some cases, I found it helpful replacing;
[external_libusb.cmake:16-18](https://github.com/IntelRealSense/librealsense/blob/master/CMake/external_libusb.cmake#L16-L18)
with
```cpp
    CMAKE_ARGS -DCMAKE_CXX_STANDARD_LIBRARIES=${CMAKE_CXX_STANDARD_LIBRARIES}
            -DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}
            -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
            -DANDROID_ABI=${ANDROID_ABI}
            -DANDROID_STL=${ANDROID_STL}
            -DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}/libusb_install
```
(and recreating the build folder)
This ensures there is no mismatch in ABI on the usb.a dependency, that often solved the error you reported. 
There should be no limitation regarding Ubuntu 16 vs Ubuntu 18.
We build Android in our CI like this [.travis.yml](https://github.com/IntelRealSense/librealsense/blob/master/.travis.yml#L79-L89).

cucrui, 2019-01-14:
> In some cases, I found it helpful replacing;
> [external_libusb.cmake:16-18](https://github.com/IntelRealSense/librealsense/blob/master/CMake/external_libusb.cmake#L16-L18)
> with
> 
> ```c++
>     CMAKE_ARGS -DCMAKE_CXX_STANDARD_LIBRARIES=${CMAKE_CXX_STANDARD_LIBRARIES}
>             -DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE}
>             -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}
>             -DANDROID_ABI=${ANDROID_ABI}
>             -DANDROID_STL=${ANDROID_STL}
>             -DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}/libusb_install
> ```
> 
> (and recreating the build folder)
> This ensures there is no mismatch in ABI on the usb.a dependency, that often solved the error you reported.
> There should be no limitation regarding Ubuntu 16 vs Ubuntu 18.
> We build Android in our CI like this [.travis.yml](https://github.com/IntelRealSense/librealsense/blob/master/.travis.yml#L79-L89).

I'll try it, thanks a lot!
And I have another question, My graduation design is about Point Cloud compression and real-time compression,  and I combine Intelrealsense D435 with PCL to do compress, Is there any good instructions to make it. I have successfully implemented compression on ubuntu system PC by using PCL, now I want to do the same thing in android for real-time transmission and moving.
If you have any suggestions, I will be grateful!

dorodnic, 2019-01-14:
Great, 
I did not try to build PCL for Android.
I'd recommend to acquire depth only (Z16) and doing all calculations using RenderScript, otherwise, it might be too slow. 
Actually, I was wondering how did you install PCL on Ubuntu 18?
I had problems with apt-get from the official PPA. Did you end up building it from source?

RealSenseCustomerSupport, 2019-01-18:
@cucrui Could you please have a try with RenderScript as dorodnic suggested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-25:
@cuicrui How's it going? Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:06 PM

@cuicrui Any update from your side? Please update. Thanks!

closed 2019-02-15 by RealSenseCustomerSupport