realsenseai/librealsense #3062: No module named pyrealsense2

ShadowHunter0510, 2019-01-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version | Ubuntu 16 | 
| Platform                                 | Raspberry Pi |

### Issue Description

I am currently using the Python Wrapper to do all my programming for my RealSense camera, but I’ve been having trouble importing the pyrealsense2 library into python. I’ve tried following some old issues to troubleshoot my problem, but I’ve had no success.

- I have tried rebuilding the Python Wrapper with the -DPYTHON_EXECUTABLE flag added onto the CMAKE command. I found the proper python executable by printing the sys.executable. It still gave me the same error even after reboot. The given path is /usr/bin/python3

- I have also tried updating the PYTHONPATH environment variable and exporting the path to the path stated in the instructions. I have also tried setting the path to the path stated above.

dorodnic, 2019-01-14:
Hi @ShadowHunter0510 
Did you try the pip version?
`pip install pyrealsense2`

ShadowHunter0510, 2019-01-14:
Hi @dorodnic 

I received an exception when I tried running that command.

[image]

dorodnic, 2019-01-14:
Right, Raspberry Pi, sorry. Pip wont work. 
So you tried `sudo make install` and `export PYTHONPATH=$PYTHONPATH:/usr/local/lib`?
`cmake .. -DBUILD_PYTHON_BINDINGS=true -DBUILD_SHARED_LIBS=false`
Did you get any errors during cmake / compilation?
Do you see `pyrealsense2.so` generated?

ShadowHunter0510, 2019-01-15:
Hi @dorodnic 

Under /librealsense/build/wrappers/python, I see a couple of files with that .so file generated. There were no errors during the compilation.

[image]

ShadowHunter0510, 2019-01-16:
Hi @dorodnic 

As you’re looking for a possible solution, can you clarify what the fifth step of building the Python Wrapper means? Quote: “Alternatively, copy the build output (librealsense2.so and pyrealsense2.so) next to your script.” From the picture above, you can see that I don’t have a file named exactly pyrealsense2.so, but other versions. However, I do have a librealsense2.so file.

lramati, 2019-01-16:
The instructions as written are a little ambiguous as Python 3 defines slightly different filenames for its modules than Python 2. You want the pyrealsense2.cpython-35m-....so files instead.

ShadowHunter0510, 2019-01-16:
Hi @lramati 
Thanks for the file clarification! Can you also explain how to “copy the build output next to your script”? I’m kind of a novice at this and I’m not sure what that means.

lramati, 2019-01-16:
Just copy/paste the files. If your program is located at `~/Projects/librealsense_project.py` you'd want to copy those files so that they're also in the `~/Projects/` folder

ShadowHunter0510, 2019-01-16:
Hi @lramati 
So if I have all my .py files in one folder, I just have to copy and paste those files into the same folder? I’ve been having trouble importing the pyrealsense2 library into my code.

lramati, 2019-01-16:
exactly. Those files are the implementation of pyrealsense2, so they have to be somewhere that python can find them. That can either be done by pointing python at the right folder (Step 4), or by putting them in the same folder as the code you're running (Step 5)

RealSenseCustomerSupport, 2019-01-19:
Hello @ShadowHunter0510,

Sorry to hear you are having trouble with pyrealsense2 compilation.

You mention you are compiling with path, "/usr/bin/python3".

Do you think you can:

-  Try compiling with given path "/usr/bin/python"?
-  After compilation with Python2, update PYTHONPATH with command:

- `export PYTHONPATH=$PYTHONPATH:/usr/local/lib`
-  After export command, copy the newly built files "librealsense.so" and "pyrealsense.so" into the folder of the .py scripts you want to run.
-  Type "python --version" and verify the version of python you are running will be Python 2.7.
-  Run python scripts with "python scriptname.py".

Please update this ticket with your results.
Thank you!

ShadowHunter0510, 2019-01-22:
Hi @lramati 
Sorry for the delay. I tried copying and pasting those files into the same folder where I store all of my .py files, but I get the same import error. I will try following the procedure as stated in @RealSenseCustomerSupport once again and see if anything changes.

lramati, 2019-01-22:
Looking back at your screenshots, it indeed looks like you were trying to
use Python 2. Following their instructions and being sure to force
compilation of Python 2 libraries should work, or alternatively, you can
use the Python 3 libraries you compiled by typing `python3 script_name.py`.
However, it's worth pointing out Python 3's syntax is slightly different in
some regards so the code might need some minor porting work

ShadowHunter0510, 2019-01-22:
@lramati and @RealSenseCustomerSupport 
I’ve tried the instructions as followed to no avail. Do you have any other suggestions?

lramati, 2019-01-22:
Can you share the command you're using to configure cmake?

ShadowHunter0510, 2019-01-22:
Hi @lramati

Most recently, I added the flags  -DBUILD_PYTHON_BINDINGS=true, -DBUILD_SHARED_LIBS=false, and -DPYTHON_EXECUTABLE=/usr/bin/python as stated by @dorodnic and @RealSenseCustomerSupport. As I said before, I realized that the code successfully compiles on the terminal but no on an IDE. I’m using Spyder at the moment for my Python needs.

ShadowHunter0510, 2019-01-24:
Hi @lramati 

After rebooting my Raspberry Pi, the Python program would not compile once again. However, after exporting the Python path to the correct one, it would successfully run in the terminal. Is there anyway to permanently export the path without having to enter this line of code every time upon starting the computer?

lramati, 2019-01-24:
Add it to the end of your bashrc file (Or the analogous one for whichever
shell you're using) there should be plenty of detailed examples floating on
the internet if you need help with that

RealSenseCustomerSupport, 2019-01-25:
Hello again @ShadowHunter0510,

You said you are using spyder for your python needs.

When you installed pyrealsense2, did you use "pip", or "conda"?

If you used "pip" install while using spyder IDE it may have broken the pyrealsense2 module install.

Please take a look at this documentation in regards to spyder python dev environment and package installation:
https://github.com/spyder-ide/spyder/wiki/Working-with-packages-and-environments-in-Spyder

You can try:
-  If you haven't already, instead of using "pip" for install, use "conda" to install pyrealsense2 (I am not familiar with spyder, this may not work).

-  Reinstalling Ubuntu, do not install spyder IDE, and just use native python libraries that are included with Ubuntu.

Please update this ticket with your results.

Thank you!

RealSenseCustomerSupport, 2019-01-30:
----------------------------------------------

Nate Ward, Jan 30, 10:40 AM

Close issue, no response given in 5 days.

closed 2019-02-04 by dorodnic

juliahambright, 2021-02-07:
Hi,

I am having a similar issue to this. I have tried the new cmake command and then export after and I still get "nomodule pyrealsense2." I have the pyrealsense2.so in /librealsense/build/wrappers/python so I tried moving it to the same folder where I am running my python script, but when I move it, it changes colors (I am on Ubuntu 18.04). When it is in the wrappers/python directory it is a light blue color, and when I move it it turns red, and gives the same nomodule error.

liamwhittle, 2021-02-09:
I second that. Have exported path, installed the .so files in /usr/local/libb, done everything, but python2.7 doesn't seem to recognize the import. Tried using Jetsonhack's script for building realsenseSDK

liamwhittle, 2021-02-09:
As per https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
you need to add the flag -DPYTHON_EXECUTABLE=[full path to the exact python], which in my case was /usr/local/lib/python2.7. However despite doing this I am still only able to import to python3, not python2

juliahambright, 2021-02-09:
Okay so mine only works in python 2 now and when I try python 3 I get a segmentation fault. Maybe I need to try /usr/local/lib/python3.6 which is the python 3 that I have. 

Also the only way I got it to work was by moving everything, except for the cmakes, in /librealsense/build/wrappers/python to the folder where I was running the python code.

liamwhittle, 2021-02-09:
It works in python2? Could you please explain your installation process to get it to work with python2?

juliahambright, 2021-02-09:
Sure, in the /librealsense/build folder I did:

cmake ../ -DBUILD_PYTHON_BINDINGS=true -DBUILD_SHARED_LIBS=false -DPYTHON_EXECUTABLE=/usr/bin/python   

(the DPYTHON_EXECUTABLE is probably where I should try python3 instead - will try right now) then I ran 

make

in the same folder, then I did:

export PYTHONPATH=$PYTHONPATH:/usr/local/lib

Then I moved all of the files in /librealsense/build/wrappers/python to the directory where my python script is located, and it works when I run "python myscript.py" but not when I run "python3 myscript.py" -- but I'm trying the python3 way right now

juliahambright, 2021-02-09:
Doesn't work for python3, I still get a segmentation faul, @liamwhittle how did you get it to work in python3?

liamwhittle, 2021-02-10:
> Sure, in the /librealsense/build folder I did:
>
> cmake ../ -DBUILD_PYTHON_BINDINGS=true -DBUILD_SHARED_LIBS=false -DPYTHON_EXECUTABLE=/usr/bin/python
>
> (the DPYTHON_EXECUTABLE is probably where I [...]

The system I'm working with already had it working for python3, so I'm not 100% sure. I randomly tried compiling the branch called 54 with the above settings. It worked. Don't know why. Also tried the "legacy" branch which didn't work. I was almost ready to give up. Hopefully that works for python3.

liamwhittle, 2021-02-10:
> As per https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
> you need to add the flag -DPYTHON_EXECUTABLE=[full path to the exact python], which in my case was /usr/local/lib/python2.7. However despite doing this I am still only able to import to python3, not python2

I was totally wrong here - I wasn't giving it the path to the python2 executable, but something else. oops

chuyyy, 2021-08-19:
Hi everyone, 

I had a problem similar to this. For context I am on Ubuntu 20.04 and using python3.6. In my situation the pyrealsense files were being installed at this path:

/usr/lib/python3/dist-packages/pyrealsense2/

I didn't try to fix the installation path. Instead I manually moved the pyrealsense files to /usr/local/lib/python3.6/site-packages/ but I was still getting the error 'No module named pyrealsense2'. I noticed that the pyrealsense file names were more complex. In my case they were named: 

pyrealsense2.cpython-38-x86_64-linux-gnu.so
pyrealsense2.cpython-38-x86_64-linux-gnu.so.2.42
pyrealsense2.cpython-38-x86_64-linux-gnu.so.2.42.0

After I stripped the cpython-38-x86_64-linux-gnu from the filenames pyrealsense2 was finally able to be imported. The file names looked like this after I renamed them:

pyrealsense2.so
pyrealsense2.so.2.42
pyrealsense2.so.2.42.0

Hope this helps.

liamwhittle, 2021-08-21:
> Hi everyone,
>
> I had a problem similar to this. For context I am on Ubuntu 20.04 and using python3.6. In my situation the pyrealsense files were being installed at this path: [...]

Yeah, the installation of the .so files often isn't straight forward! Glad you got it to work - I can't remember when I managed to figure it out but it took quite some time. On another note, librealsense isn't recognizing my T265 camera... ahh the endless issues with this software

chuyyy, 2021-08-21:
> Yeah, the installation of the .so files often isn't straight forward! Glad you got it to work - I can't remember when I managed to figure it out but it took quite some time. On another note, [...]

I gave up trying to fix the install path so I just manually moved the files. The thing is that on Ubuntu 16.04 the file names were named like so - pyrealsense2.so - but on Ubuntu 20.04 they had the string cpython-38-x86_64-linux-gnu added ie pyrealsense2.cpython-38-x86_64-linux-gnu.so and that's when I started getting the 'No module named pyrealsense2' error. I'm not sure why removing the substring cpython-38-x86_64-linux-gnu fixed the error but I was happy that it worked. 

Regarding the issue that you're having with librealsense not recognizing your camera. Have you applied the udev rules?

liamwhittle, 2021-08-22:
> I gave up trying to fix the install path so I just manually moved the files. The thing is that on Ubuntu 16.04 the file names were named like so - pyrealsense2.so - but on Ubuntu 20.04 they had the [...]

yep, didn't help. Librealsense latest version works fine, but I had installed 2.40 (which is what worked with building the wrapper for ROS2 Eloquent). It's so difficult to manage the installations of librealsense - I think there are .so files floating around for like 3 different versions, but how do I know where to find them? I can't keep all the source files due to space constraints... will probably have to try again on a clean ubuntu install, bu reflashing jetsons is a huge pain...

JaeYoungLee79, 2022-02-21:
Dear Everyone

Is it solved? i'm same situation.

No module named 'realsense_depth'

FLandolt, 2023-11-16:
Hi, I have struggled with his for quite a wile now. For me, the fix at the end was that I did not have the correct paths included me adding this at the end of my .bashrc file fixed everything:
```
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/home/serveradmin/librealsense/build/Release
```
The last line, including the "Release", was missing everywhere in the descriptions and tutorials. Maybe i did something wrong with my instalation but this fixed it!
In this folder there is all the .so files:
```
~/librealsense/build/Release $ ls
librealsense2-gl.so                        
librealsense2-gl.so.2.54                         
librealsense2-gl.so.2.54.2                
librealsense2.so                  
librealsense2.so.2.54               
librealsense2.so.2.54.2                   
pybackend2.cpython-311-aarch64-linux-gnu.so      
pybackend2.cpython-311-aarch64-linux-gnu.so.2        
pybackend2.cpython-311-aarch64-linux-gnu.so.2.54.2   
pyrealsense2.cpython-311-aarch64-linux-gnu.so      
pyrealsense2.cpython-311-aarch64-linux-gnu.so.2.54  
pyrealsense2.cpython-311-aarch64-linux-gnu.so.2.54.2  
pyrsutils.cpython-311-aarch64-linux-gnu.so            
pyrsutils.cpython-311-aarch64-linux-gnu.so.2.54 
pyrsutils.cpython-311-aarch64-linux-gnu.so.2.54.2    
```
I hope this helps anyone having the same issue and struggles i had.
 (Fall 2023, Python 3.11, protobuf 3.14.0)

MartyG-RealSense, 2023-11-16:
Thanks so much @FLandolt for sharing your solution!