realsenseai/librealsense #3063: Navigation using realsense and python

MaryBD, 2019-01-14:
Do you know if there is any outdoor navigation sapmles or refrences using realsense and python?

MartyG-RealSense, 2019-01-14:
There is an RGBD SLAM navigation system for Python called RGBD_PTAM.

https://github.com/uoip/rgbd_ptam

Whilst I have not seen an example of it being used in combination with a RealSense camera, it ought to work.

MaryBD, 2019-01-15:
Thank you. I am looking for other faster online methods rather than slam, and I am also on windows.

MartyG-RealSense, 2019-01-15:
You should probably think in terms of drones and flight controllers.  There is a controller called DroneKit-Python that can be installed in Windows.

**About**
http://python.dronekit.io/about/overview.html

**Installation**
http://python.dronekit.io/contributing/developer_setup_windows.html

MaryBD, 2019-01-15:
Sounds impressive thank you. I am going to use realsense together with a mobile robot for outdoor navigation, so I am looking for something that goes with this platform.

RealSenseCustomerSupport, 2019-01-21:
@MaryBD Sorry we don't have such sample currently. If you have any good finding about this, please feel free to share here. Thanks!

RealSenseCustomerSupport, 2019-01-25:
@MaryBD Any other questions about this ticket? Looking forward to your update. Thanks!

MaryBD, 2019-01-28:
I am still working on that, and I will share if I got any. I will be happy if other people also share their open sources.

closed 2019-02-23 by RealSenseCustomerSupport