realsenseai/librealsense #3068: Unable to get exposure or gain from stream with the openni wrapper

yashmanian, 2019-01-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Linux Ubuntu 16 | 
| Platform                                 | PC |
| Language                            |  C++ | 

### Issue Description
I am trying to use the openni wrapper with the Realsense. I am having problems with getting stream properties such as gain, exposure, and checking if auto exposure or auto white balance is set. I cannot even set said parameters on the stream.

For example, when I try to set the exposure, I get the following error:
~~~
[RS2] ERROR at FILE librealsense/wrappers/openni2/src/Rs2StreamProps.cpp LINE 342 FUNC getProperty propertyId=101 dataSize=4
[RS2] ERROR at FILE librealsense/wrappers/openni2/src/Rs2StreamProps.cpp LINE 342 FUNC getProperty propertyId=100 dataSize=4
[RS2] setProperty propertyId=102 dataSize=4
[RS2] ERROR at FILE librealsense/wrappers/openni2/src/Rs2StreamProps.cpp LINE 97 FUNC setProperty propertyId=102 dataSize=4
~~~
Is there a way I can fix this issue?
Thank you.

RealSenseCustomerSupport, 2019-01-21:
@yashmanian What librealsense and FW version are you using when you get this issue?

wongfei, 2019-01-21:
@yashmanian 

the easiest way to debug this issue is patching file Rs2Base.h and checking for more detailed errors:

```
class Rs2Error
{
public:
	...
	//inline bool success() const { return m_error ? false : true; } // <-- PATCH
	inline bool success() const { if (m_error) { rsTraceError("rs2 error: %s", get_message()); } return m_error ? false : true; }
	...
```
	
or you can just trace getProperty method and see why it doesnt reach ONI_STATUS_OK:
	
```
OniStatus Rs2Stream::getProperty(int propertyId, void* data, int* dataSize)
{
...
	case ONI_STREAM_PROPERTY_AUTO_WHITE_BALANCE:
		{
			if (data && dataSize && *dataSize == sizeof(OniBool) && m_oniType == ONI_SENSOR_COLOR)
			{
				Rs2Error e;
				float value = rs2_get_option((const rs2_options*)m_sensor, RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, &e);
				if (e.success())
				{
					*((OniBool*)data) = (int)value ? true : false;
					return ONI_STATUS_OK;
				}
			}
			break;
		}
...
```

RealSenseCustomerSupport, 2019-01-25:
@yashmanian Any update about this issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:16 PM

@yashmanian Another point - have you applied the kernel patch described as https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

closed 2019-02-07 by yashmanian