realsenseai/librealsense #3070: Will D435i add support for POSE stream (as provided by T260)

daniel-packard, 2019-01-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  N/A                                      | 
| Platform                                 | PC  |
| SDK Version                            |  librealsense (RealSense2 SDK) | 
| Language                            |  C/C++                        | 
| Segment			|  Health                         | 

Hi team - we are looking into using the IMU of the D435i sensor for tracking a 6DOF pose. 

Looking at some other pull requests, it looks like the T260 module has support for RS2_FORMAT_POSE (https://github.com/IntelRealSense/librealsense/pull/2594) -- this would be perfect for us!

However, the D435i does not support the POSE stream. It _does_ support raw GYRO and ACCEL streams, which would allow us to build the pose tracking manually. This has some examples for tracking the rotation only (not translation).. 
* https://github.com/GruffyPuffy/imutest
* (and based on that) https://github.com/IntelRealSense/librealsense/pull/3053

So our question is - are there plans to enable the POSE stream for the D435i camera module?

side note: We're hoping the answer here was written before the D435i existed: https://github.com/IntelRealSense/librealsense/issues/1078

MartyG-RealSense, 2019-01-15:
Dorodnic the RealSense SDK Manager commented on December 17 2018:  "POSE stream type can be enabled in the future for the D435i using Intel RealSense SLAM or provided directly from a dedicated device. While neither are officially confirmed yet, a prototype of standalone tracking device was showcased at this years iROS. There is some public information at our site".

https://realsense.intel.com/solutions/#virtual-reality

daniel-packard, 2019-01-15:
@MartyG-RealSense - Thanks for the quick response!

When you say "support in the software sensor was added" does that mean it should be supported by the actual D435i device? Or just that the hooks are there, and that per @dorodnic's commment it "can be enabled in the future"? 

I need to read into the `software_sensor` interface a bit to understand.

Can you link the original comment above by @dorodnic with more context?

thanks!

MartyG-RealSense, 2019-01-15:
Dorodnic or one of the other Intel guys on this GitHub can give a better explanation than I can.  *smiles*

The Dorodnic quote came from this discussion:

https://github.com/IntelRealSense/librealsense/issues/2718#issuecomment-447767815

daniel-packard, 2019-01-15:
I understand the software device better after looking at this example: https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

Will wait to hear back from the intel guy's regarding plans for adding D435i support for POSE stream.

Thanks!

dorodnic, 2019-01-16:
Hi @daniel-packard 
We are still evaluating the possibility to provide SLAM package over D435i. 
There are several things complicating this:
1. Limited RGB FOV of the D435i is a problem for SLAM performance. You certainly can run SLAM on the D435i, but if RealSense will publish a dedicated SLAM package, we want to make sure it will work **well**.
One possible mitigation for this is using the IR sensor for tracking (with wider FOV). This cannot be done when camera projector is enabled (since the patter will confuse tracking). We are also playing with an option to run both "side-by-side" at 60 FPS (#3066) - passive frames would be used for tracking and active for depth, both at 30 FPS. Even with wider IR FOV, it still might not be enough.
2. Intel SLAM algorithm is considerably more complex than anything else in this SDK. Without dedicated optimizations, we are not sure how well it will perform on non-Intel hardware. At the same time, many of our customer are using non-Intel platforms for robotics. Distribution of such software package can also be challenging. 

So, for now, we can only provide the D435i with its hardware capabilities. With this SDK we will continue to bride the gap by adding more examples and publishing tutorials for integration with open-source SLAM packages.

daniel-packard, 2019-01-16:
Thanks for the great explanation! 

So if I understand correctly, the POSE stream provided by the T260 is not "simply" dead reckoning from integrating/filtering the raw IMU data. It includes a feedback loop with visual tracking from a fisheye video stream. 

We will go ahead and move forward with a dead reckoning approach based on the raw IMU streams, and validate that against our current visual odometry estimates. (Using https://github.com/IntelRealSense/librealsense/pull/3053 as a starting point)

closed 2019-01-16 by daniel-packard

dorodnic, 2019-01-23:
More information on the T265 is now available (#3129)