realsenseai/librealsense #3072: Where the actual pointcloud origin is?

yyxr75, 2019-01-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi there,
I have a question about the true origin of the pointcloud which was stored by the realsense-viewer.
I tried to figure out which camera center the pointcloud actual locate in, so I use the intrinsic matrix
> 954.319092, 0, 639.966003
0, 954.319092, 363.405853
0, 0, 1

to project the points into the image plane. But the result showed no images could match. There are not only a scale mistakes, but also position errors. 
Where the exactly pointcloud origin is? Or in other words, how does lib-realsense transfer depth frame to points?
These are some experimental result showed below:
[image]
I can see the depth image is using the left infrared camera center as its own center. The projection shows the pointcloud origin might be in the middle of the two cameras and a little bit below the baseline. But what is it's scale factor? If what I used is a correct intrinsic matrix, the calibration board should at least be the same size.
With much appreciation！

yyxr75, 2019-01-16:
> Hello @yyxr75 ,
> The depth origin is defined in [D400 spec](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf) - it is the front of the lens of the left IR imager .
> One of the nice properties derived from it is that depth and left IR are perfectly aligned.
> Similar to #3009

Hello, ev-mp
I am agree with you about the depth-frame location, it is aligned with the left camera. But what confuse me is why my pointcloud could not match with any of them, Can you distinguish my the blue part of each picture, it's the region I directly projected the pointcloud.ply file into the image plane, this result shows the 3D pointcloud file which was computed by the 'getdistance(depth-frame)' function doesn't locate in any of the cameras centers. 

WHERE EXACTLY IS MY POINTS 'PLY' FILE's ORIGIN? Do you have any ideas?

ev-mp, 2019-01-16:
@yyxr75 , it is not very clear from the description what each image represents and what is the discrepancy you see. Please elaborate.

With regards to data coverage in Depth vs Left/Right IR sensors:
The 3d points are derived from the depth frame.
The depth frame, in turn, is based on data from left and right IR images found in overlapping areas. The areas that appear exclusively in the view frustum of either Left or Right (IR) sensors will not have depth data generated, e.g. the leftest columns of pixels in left IR and rightest columns in right IR images.
Using the sensor intrinsic you may obtain the principal point that represents the focal center in pixel's coordinates.
The PLY origin is identical to depth frame origin as specified above with X axis pointing right, Y - down, and the Z axis that passes along the principal point and pointing forward.

yyxr75, 2019-01-16:
> @yyxr75 , it is not very clear from the description what each image represents and what is the discrepancy you see. Please elaborate.
>
> With regards to data coverage in Depth vs Left/Right IR [...]

Sorry I didn't make a clear statement.

The main idea of my experiment is to figure out the 3D pointcloud's origin in PLY file and a orientation matrix of them. Like you just said, the 3D points are computed from the depth image, so I could presume the 3D points and the depth image owns the same origin points (0,0,0). 

I would try my best to describe my experiment steps clearly below.

First of all, I use the D-415 to capture a calibration board like below:
[image]
I stored these images one by one, depth image, left infrared image, right infrared image, then the color image. And then, I change the viewer to the 3D display showed below and click the save button to save the 3D ply file.
[image]
If just as the normal knowledge describes, the 3D pointcloud are computed from the depth image, I can just use the depth intrinsic matrix to project the points into depth image plane and the region of calibration board in depth image will perfectly covered by the projecting points in image plane. Am I right of this?
So I use MATLAB to read the ply file and do the projection procedure. I filtered the points and only choose the points on the board like below:
[image]
Then I got the location of every points like (x,y,z) and a intrinsic matrix of depth camera as below:

> 954.319092, 0, 639.966003
0, 954.319092, 363.405853
0, 0, 1

Then I calculate the pixel location corresponding to the 3D points in the depth image (also in the other two images) using the equation:

> s(u,v,1)' = [954.319092, 0, 639.966003
0, 954.319092, 363.405853
0, 0, 1]*(x,y,z)'

Finally, I plotted the projection result along with the depth image together and see how much shift exists. Ideally, the (u,v) pixel region will perfectly cover the board region of depth image, but the consequence didn't appear like this. I can see the depth image is perfectly aligned with the left infrared camera, but the projection blue region can't cover the depth calibration board region with some scale shift and position shift visible to the naked eye.
[image]
Is this means the origin of 3D points not with the depth image? Or was the coordinate of the 3D points transfered by any of my processing procedure?
I hope I have made my whole idea clearly this time and wish for any advices with my best appreciation!

lramati, 2019-01-16:
Are you remembering to divide through by s? I recently had issues because I forgot to do this in a project unrelated to realsense. Just taking the U, V values without first dividing by the 3rd value in the vector could explain the scale error

yyxr75, 2019-01-16:
Yes sir
I divided the scale factor s using this equation
(x’,y’,z’)=intrinsic matrix*(x,y,z)
And then divide the s using this
u=x’/z’
v=y’/z’
Am I right with this?

lramati, 2019-01-16:
Yep

yyxr75, 2019-01-16:
If I am not wrong, I just can’t go any longer with this shit! Would u mind
to check this out for me ? I will email my matlab code and data to you
tomorrow (after 8hours maybe because I am here midnight in China and I am
on my bed now sorry) if you are willing to help. So much appreciation!

yyxr75, 2019-01-17:
Sir,
I tried the procedure just now, and find it was align with the left infrared camera. How a stupid fault I have made I don't even know what happened to this shit. But result turns out good whatever.
Sorry bother you for such a long time.

closed 2019-01-17 by yyxr75