realsenseai/librealsense #3073: box_dimensioner_multicam__demo.py，Doesn't the camera already have internal and external references? Why do we have to calibrate the checkerboard?

honghande, 2019-01-16:


dorodnic, 2019-01-16:
I'm not sure what you mean by internal and external references. The camera does not track its position, and solving general point-cloud matching problem is challenging when angles are wide, so some calibration is required.

honghande, 2019-01-17:
hi，dorodnic 
What I mean by the camera's internal and external parameters is the data of Intel calibration. Can't we use these parameters when making measurements? Aren't the depth information of binocular measurement just distance information? These distances are available. Why do we need to calibrate them separately?

RealSenseCustomerSupport, 2019-01-21:
Hi @honghande,

As the description of the demo example page, this sample demonstrates the ability to use the SDK for aligning multiple devices to a unified co-ordinate system in world to solve a simple task such as dimension calculation of a box. Optimal rotation between two co-ordinates using Kabsch Algorithm. The realsense camera's intrsinic and extrinsic parameters are used in the example.

You can also refer to "rs-measure" example to create your own box measurement app.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

honghande, 2019-01-22:
Hi，@RealSenseCustom
The checkerboard calibration is mainly for coordinate conversion of two depth cameras, right? A single depth camera does not need checkerboard to calibrate it? Can we use the internal and external parameters calibrated by the depth camera for measurement?

RealSenseCustomerSupport, 2019-01-23:
Hi @honghande,

D400 camera with its own intrinsic and extrinsic parameters can be used to have the measurement.  The demo example can be modified to achieve that.
But, you will see IR dots on the image since the emitter is enabled for better depth.

closed 2019-03-04 by RealSenseCustomerSupport