realsenseai/librealsense #3074: Texture explanation - Realsense D435

spaltiel, 2019-01-16:
Dear All,

1. Alignment between point cloud texture: 

We reviewed and use the example (librealsense/examples/align/rs-align.cpp). In addtion we also use the following (form another example) in order to match point cloud with texture:

std::tuple<uint8_t, uint8_t, uint8_t> get_texcolor(video_frame texture, texture_coordinate texcoords)
		{
			const int w = texture.get_width(), h = texture.get_height();
			int x = std::min(std::max(int(texcoords.u*w + .5f), 0), w - 1);
			int y = std::min(std::max(int(texcoords.v*h + .5f), 0), h - 1);
			int idx = x*texture.get_bytes_per_pixel() + y*texture.get_stride_in_bytes();
			const auto texture_data = reinterpret_cast<const uint8_t*>(texture.get_data());
			return std::tuple<uint8_t, uint8_t, uint8_t>(
				texture_data[idx], texture_data[idx + 1], texture_data[idx + 2]);
		}

Is it possible to get for each frame and for each point inside the points cloud it texture without the need to iterate over all the points? or can you recommend a faster way other then 'get_texcolor'?

2. Documantations
Writing the following:
...
rs2::video_frame color = frames.get_color_frame();
...
const auto New_Texture = reinterpret_cast<const uint8_t*>(color.get_data());
....

We figured that New_Texture should hold the 2D image which we mapped. Indeed, we retrieve it, however, do you have docs which explains what kind of data each object contain (in above example we could not figure out in advanced what color.get_data() will return nor it expected size)? 

10x!,
Amir I.

ev-mp, 2019-01-17:
Hello @spaltiel,
It would be more clear if you could describe the use-case you're trying to achieve, in order to adjust the answers. 

1) 
 Finding the color (RGB tupple) of a single depth pixel shall be performed as follows:

   (1)  Convert 2d pixel into 3d coordinate  - `rs2_deproject_pixel_to_point(..).`.
   (2) Transform the 3d coordinate from depth sensor coordinates system into RGB sensor's  - `rs2_transform_point_to_point(...)`.
   (3) Calculate the 2D rgb pixel coordinate that corresponds to the 3d coordinate - `rs2_project_point_to_pixel(...)`.
   (4) Get the RGB value of that pixel coordinate.

Following the example mentioned, you should follow the [example you mention](https://github.com/IntelRealSense/librealsense/blob/development/src/proc/align.cpp#L42):
> **Input**: __depth_point__   - a valid (value >0) pixel in the depth image.
 rs2_deproject_pixel_to_point(depth_point, &depth_intrin, depth_pixel, depth);
 rs2_transform_point_to_point(other_point, &depth_to_other, depth_point);
 rs2_project_point_to_pixel(other_pixel, &other_intrin, other_point);
    std::tuple<uint8_t, uint8_t, uint8_t>__ get_texcolor(video_frame texture, texture_coordinate texcoords
  **Output**: __RGB pixel__  value.

2)
>const auto New_Texture = reinterpret_cast<const uint8_t*>(color.get_data());

1) `color.get_data()` - provides you an direct access to the pixels stored in the `color` object.
The content is dependent of the format negotiated on stream start (RGB/BGR/YUYV).
In case of RGB/BGR the internal alignment is standard 3-bytes per pixel (3 eight-bit RGB components) arranged row-wise.
For YUYV it is a two-bytes per pixel according to [YUYV spec](https://www.linuxtv.org/downloads/v4l-dvb-apis-old/V4L2-PIX-FMT-YUYV.html)

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

abernste, 2019-01-20:
@spaltiel - 
is ev-mp explanation helpful? 
10x

spaltiel, 2019-01-20:
Hi, 

We don't think it solve our issue (we hope that indeed we understood the answer correctly :) ).

Our case:
We would like to forward the point cloud to C# (XYZ+ RGB info for each pixel).
what we already did:
We get info (X,Y,Z and RGB data) for each point and crated new method according the example we mentioned and build our own struct which we will transfer to C#. However it is a slow process. Therefore, we try to figure if there is a better (faster) way to do it ? 
2. From this issue we learned that we don't understand how to figure the data structure of the objects as an example we we mentioned the color object. Therefore we asked for the documentation of the data structures if you can share it.

BTW the origin to this problem was that we could not get vertexes and RGB in C# directly using the current wrapper (similar to the way get get it in C++).

We hope that we did not miss anything.

Thanks in advance!
Amir I.

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi spaltiel,

Wonder if you have any updates to share or do you still need help?

LRS is an open source project, and all the documents and source code are shared in the github.
And as ev-mp mentioned, the basic way to is done in the example https://github.com/IntelRealSense/librealsense/blob/development/src/proc/align.cpp#L42
Based on your scenario and usage, you may develop some heuristic/optimal approach to get better/faster performance (again based on your usage to justify).

Thanks!

spaltiel, 2019-03-03:
Dear friend,

Thanks for not forgetting us!
I think that we manged to understand and manipulate the texture
sufficiently.

BTW if you have samples which use VTK we will appreciate it.

Best Regards,
A.I.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi spaltiel,

Thanks for the updates and feedback.
Glad to know that you have a working solution. Wonder if you would like share more details on your final solution?

Please check back later if existing ones don't help enough, as we are keeping adding new features and samples to RS.

Cheers!

closed 2019-03-19 by RealSenseCustomerSupport