realsenseai/librealsense #3075: Hardware Timestamp Unavailable for D415

mjfoo21, 2019-01-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 2.17.1 | 
| Operating System & Version |   Win10  |

I am unable to get the timestamp for both depth and colour cameras. The RealSense Viewer gives me this error:

> Hardware Timestamp unavailable! This is often an indication of improperly applied Kernel patch. Please refer to installation.md for more information

I have checked this [question](https://forums.intel.com/s/question/0D50P0000490SCnSAM/realsense-d435-timestamps?language=en_US) which directs to the said [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) that only contains instructions for Linux. Is there anyway to solve this for WIndows 10?

I tried to get the timestamp using the source code as well. Results are the same.

ev-mp, 2019-01-16:
@mjfoo21 ,
The HW timestamps are part of UVC metadata attributes. You can find the documentation [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md).
More specifically, in order to enable this feature in on Windows OS you can find instructions in this [link](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows).

mjfoo21, 2019-01-17:
Thank you so much! Updating registry using the instructions solved the problem. I didn't use the two "frame_metadata" functions as suggested by the link. Just `.get_timestamp()` is enough to get the timestamp.

closed 2019-01-17 by mjfoo21

Dealerpriest, 2019-04-23:
Can't find the script folder in the SDK folder. Where is it supposed to be?
I managed to copy the powershell script from github though.