realsenseai/librealsense #3076: Noise removal

EmmanuelCos, 2019-01-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 |       
| Language                            |  C++  | 

Hello,
I need to take a point Cloud picture in .ply format of a static scene, and want to have as least noise as possible.
Should I take several 3D .ply pictures, and average them or is there any better way to do this?

Thank you for your help,

Emmanuel

dorodnic, 2019-01-16:
Hi @EmmanuelCos 
The SDK includes [temporal filter](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#temporal-filter) that does some type of smart median filtering. You could pass the input through this filter (depending on the language, you can see post-processing example or post-processing notebook) and then generate point-cloud and save to PLY.

EmmanuelCos, 2019-01-16:
Thank you for the quick answer Dorodnic.
Do you know what if the behaviour by default if we don't set any option with the set_option? Is there a documentation for each option we can set?
Thank you,
Emmanuel

dorodnic, 2019-01-16:
I think its part of the link above

closed 2019-02-07 by RealSenseCustomerSupport