realsenseai/librealsense #3077: MATLAB wrapper align.m class name conflict

AndrewFielding, 2019-01-16:
The align.m in the MATLAB wrapper has a conflict with a MATLAB built in function align. Is it possible to rename this to avoid this conflict?

lramati, 2019-01-16:
My understanding is was that this is fine because the class sits inside the
realsense namespace. Did I misunderstand Matlab's namespace scoping
mechanism?

AndrewFielding, 2019-01-16:
I tried using the MATLAB align function ( I am building a simple acquistion app using the MATLAB realsense wrapper toolkit) and got the following error message: 

I'm assuming it's a conflict

Error using align
Error: File: align.m Line: 9 Column: 35
A constructor call to superclass realsense.filter appears after the object is used, or after a return.

Error in RS_acquire_gui (line 61)
align(hAcquireButton,hExitButton,'center');

I also get the following when starting MATLAB:

Warning: Function filter has the same name as a MATLAB builtin. We suggest you
rename the function to avoid a potential name conflict. 
Warning: Function format has the same name as a MATLAB builtin. We suggest you
rename the function to avoid a potential name conflict.

lramati, 2019-01-16:
Regarding the first error, that was recently fixed in PR #3069. It's a very minor change to line 8 in align.m. Could you make this change locally and tell me if that fixes things?

AndrewFielding, 2019-01-17:
Hi Lior,

The update to the align.m function seems to have fixed things. Many thanks.

Andrew

RealSenseCustomerSupport, 2019-01-24:
Hi AndrewFielding,

Glad to hear that the PR #3069 that Lior pointed out works for you.
If there is nothing else, we are going to close this one.

Thanks!

closed 2019-02-01 by RealSenseCustomerSupport