realsenseai/librealsense #3080: Matlab get_extrinsics_to error

delucr, 2019-01-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (0.5.10.13.00) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {Matlab}                          | 
| Segment			|  {Other}                          | 

### Issue Description
I am trying to get the extrinsics properties from IR1 to IR2 in Matlab  (R2016b).  I followed the API how to for getting the disparity baseline (https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-disparity-baseline).  My code looks like this:

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();

config = realsense.config();
config.enable_all_streams();

profile = pipe.start(config);

ir1_stream = profile.get_stream(realsense.stream.infrared,1)
ir2_stream = profile.get_stream(realsense.stream.infrared,2)

ir1_stream.get_extrinsics_to(ir2_stream)

I get the following error:
Error using librealsense_mex
rs2::stream_profile::get_extrinsics_to: Wrong number of inputs

Error in realsense.stream_profile/get_extrinsics_to (line 80)
            extrinsics = realsense.librealsense_mex('rs2::stream_profile', 'get_extrinsics_to', this.objectHandle, to.objectHandle);

This seems to be a problem with the mex wrapper.  Thoughts?

RealSenseCustomerSupport, 2019-01-25:
@delucr Did you get it through? Looking forward to your update. Thanks!

delucr, 2019-01-25:
> @delucr Did you get it through? Looking forward to your update. Thanks!

Not yet. I've been using the Windows Installer version of the library and I could not get the mex code to compile correctly the first time I tried, so I set it aside for now.  I'll let you know if I get it to compile or once the change has been merged in the installer.

delucr, 2019-01-25:
@RealSenseCustomerSupport,  tested it this morning and it works.

RealSenseCustomerSupport, 2019-01-28:
@delucr Thanks for your update! That's great! Close the ticket accordingly.

closed 2019-01-28 by delucr

nevynwong, 2019-05-29:
@delucr ，I would like to ask how to get the camera parameters, because I need these parameters to convert the depth to point cloud. I refer to your method, but I don't know how to solve the error you mentioned. I am worried now.Can you help me?

delucr, 2019-05-29:
@nevynwong, if you use the latest release of the SDK and update Matlab library, you shouldn't get the error.  If you are using the old code you can fix the error yourself by editing  line 113 of librealsense_mex.cpp as mentioned in the above thread.