realsenseai/librealsense #3082: Adjust latency video stream latency in unity?

imanjj, 2019-01-16:
Hi,

I'm using D435 in Unity Windows. I want to change the latency of the video stream to measure my performance. I'm a beginner. Can you help me with some guidance?

Thanks
Best,
Iman

dorodnic, 2019-01-17:
Hi @imanjj 
This is somewhat general question. I don't know if there is some Unity specific method you could try, but I would create C# `rs2::frame_queue` object. Then on every new frame:
* q.enqueue(f)
* if (q.size() > 100) f = q.dequeue(); process(f)
This will introduce 3 sec latency (100 frames x 30 ms between frames)

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

imanjj, 2019-01-18:
Thanks @dorodnic! 
Indeed queuing worked but I can't make it work with qsize bigger than 15! I'm using the hello-world sample. The program crashes after waiting for a few seconds. I have attached the code below

Best,

`#include <iostream>

#include <librealsense2/rs.hpp>

using namespace rs2;

void main()
{
    // Create a Pipeline, which serves as a top-level API for streaming and processing frames
    pipeline p;

    // Configure and start the pipeline
    p.start();
	int i = -1;
	const int n = 16;
	frame_queue* q = new frame_queue(n);
	
    while (true)
    {
		i++;
		// Block program until frames arrive
        frameset frames = p.wait_for_frames();
		q->enqueue(frames);
		if (i < n) continue;
		frames = q->wait_for_frame();
		
		// Try to get a frame of a depth image
		depth_frame depth = frames.get_depth_frame();
		// The frameset might not contain a depth frame, if so continue until it does
		if (!depth) continue;
		
		float width = depth.get_width();
        float height = depth.get_height();

        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(width / 2, height / 2);

        // Print the distance 
        std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
    }
}`

dorodnic, 2019-01-23:
Hi @imanjj 
By default, when we detect >16 frames in use, we stop sending new frames to let the App handle the load, and avoid introducing unlimited latency and memory increase.
To bypass 16 frames limitation, you could call `.keep()` method on `frame` object you are delaying.

RealSenseCustomerSupport, 2019-01-24:
Hi imanjj,

Did you try the suggestion from Dorodnic? Any updates?

Thanks!

imanjj, 2019-01-24:
Hi,

I haven't gotten to try it yet. I will respond as soon as I tried it. Thanks for the great support!

Best,
Iman

RealSenseCustomerSupport, 2019-02-04:
----------------------------------------------

Hi imanjj,

Wonder if there is any update?

Thanks!

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi imanjj,

Not sure if you have tried it? Or if you still need help on this one?
Otherwise, it will be closed soon.

Thanks!

imanjj, 2019-02-14:
Hi @dorodnic ,
The object that I'm trying to keep is frameset. I only see that I have access to videoFrame or DepthFrame from the frameset. Even when I get to video or depth frame object I don't see a keep function.

Best,
Iman

closed 2019-02-20 by RealSenseCustomerSupport

ogoshen, 2019-02-24:
PR #3285 brought back the `Frame.Keep` method.
To keep the entire frameset you should be good to call `Keep` on the frame returned from `FrameSet.AsFrame`