realsenseai/librealsense #3084: HW Sum

pncanas333, 2019-01-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 and D415 } | 
| Firmware Version                   | (5.11.1) | 
| Operating System & Version |   Win (10)   | 
| SDK Version                            |  { 2.17.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/python }                          | 
| Segment			|  {others}                          | 

### Issue Description
Hi,

I have three Realsense cameras (1 D435, 2 D415) and I want to do a HW synchronization between them. I have read the document: https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf, I have also read documentation and many forums. There is some confusion for me, among all that information. I want to know, how can I do the HW synchronization and what do I need? This, because:

- I just ordered the cables and the things that the paper mentioned to do the HW synchronization. But then I read that there is no need for that ...
- So I tried to synchronize through RealSense-Viewer, with the Sync button. It says: enable sequence synchronization ... but the button seems to be disabled. (I do not know if I'm missing something here or if I need the wiring) I assigned the role of master to a camera and the role of slaves to others, disabled the auto exposure, set the fps to 30 and it did not work either. Each camera has a different USB port on the same computer.
- Also, it´s been a little while since other people asked questions about HW Sync on the forums. Maybe in that time Intel has done something new and easier... I hope :´(  

The thing is, I just want a clarification on this topic. I have not tried to physically connect the cameras because I'm still waiting for the products to arrive, but I just want to make sure what is the right thing to do and what will work. Also, the goal of all this is to make all the three cameras start and stop recording at the same time. If there is another way to do this, please tell me.

Thank you so much for reading, I appreciate any suggestion :)

vladsterz, 2019-01-17:
Hello @pncanas333 .
In order to synchronize in the hardware sense, you need the cables. Why is that, I hear you ask. When a device is set to **master** state, whenever it acquires a frame (or just before its acquisition, don't know), it also generates a electrical pulse. Also, whenever a device is set to **slave** state, a pulse at its input (pins 5 and 9 if I remember correctly) makes it to **try** to acquire a frame. (If it fails to acquire a frame at the moment of the synchronization pulse, is still will acquire a frame before the next synchronization pulse). So yes, you need the cables for the HW sync.
As our experiments show( we haven't tested it thoroughly) that it works as intended.

Dancho80, 2019-01-17:
[Please Ignore - RealSense system comment]

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

dorodnic, 2019-01-17:
Hi @pncanas333 
Sorry if the UI is a bit confusing. 
Hardware sync (setting camera to master / slave) is one of the options in Viewer, but it is under Stereo Module > Controls.
"Software Sync" grayed out button (or the small lock button in the 3D viewer) controls if librealsense will try to bundle frames at software level. When hardware sync is enabled, it is not recommended, since every camera has different clock.
You might ask how then to bundle related frames in code. The whitepaper recommends to just use closest frame-sets in time of arrival. This seem to work best.
However, there is a separate discussion about accessing the hardware clock (#2922) to sync it to host clock. We are still working on that, but you have a code snippet to try.

pncanas333, 2019-01-18:
Okey, thank you for your answers. I will try the HW sync connecting the cables! Wish me luck 💃 

Quick suggestion for Intel: Do video tutorials please. I know there are already some videos but if you could do something more specific and proyect oriented like: HW sync realsense cameras for .. (whaterver), it would be great :) Some people (including me) don´t know exactly all the things you can do with realsense cameras. 

Ty, have a nice day.

HippoEug, 2019-01-18:
@pncanas333 The video tutorial suggestion is a good one!! I remember getting the camera for the first time few months ago, didn't even know how to use the SDK! Tried searching everywhere for video tutorials initially, couldn't find any except couple videos showcasing some cool projects.. was frustrating initially for beginners like myself.

RealSenseCustomerSupport, 2019-02-06:
----------------------------------------------

Will try and see how to deliver the video tutorial. Thanks.

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

Currently work in progress on the video capture by using screen recorder. Will publish the links about where to download the tutorial video.

Thanks

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

Hi

The "How-To" tutorial video is just published to Youtube.
Please read the description of each video clip:
https://youtu.be/8pT9xYIiXqw
https://youtu.be/0bMTpm_CMrI
https://youtu.be/DSm7NYG5bZE

Your comments are highly appreciated.

Thank you

closed 2019-03-01 by RealSenseCustomerSupport

szx0112, 2020-02-04:
@RealSenseCustomerSupport Thanks for the videos. I follow the 3rd tutorial video for multicamera sync on two d435i cameras. I also connected a third camera without sync for comparison. However, I could not find significant difference on video fps, it seems that the depth stream of all three cameras are pretty close (30HZ). Could you provide a quick video for d435i as well?