realsenseai/librealsense #3086: How can we have multiple object detection using MobileNetSSD?

MaryBD, 2019-01-17:
I am following the tutorial on object detection using realsense and MobileNetSSD. AS a result one object on the screen is detected and shown by a rectangle around it. Do you know how I can show two or more objects that exist in the frame simultanously? I use python 3.6, win 10, realsense D345. 

#python

RealSenseCustomerSupport, 2019-01-17:
[Please Ignore - RealSense system comment]

MaryBD, 2019-01-17:
Thanks for your quick reply. Do you know if there is such a code in python?

MartyG-RealSense, 2019-01-17:
There is a list of Mobilenet SSD programs in Python format at the link below.

https://github.com/topics/mobilenet-ssd?l=python

Of that list, the images in the links below suggest the ability to do multi-object recognition.

https://github.com/apoorvavinod/Real_time_Object_detection_and_tracking

https://github.com/PINTO0309/MobileNet-SSD

MaryBD, 2019-01-17:
Thank you very useful. Just do you know how I can give value to output config , ... I am using pycharm and it returned error on any try for giving value to the required arguments.

MartyG-RealSense, 2019-01-17:
I will leave this question to Dorodnic or one of the other Intel guys, as they can answer a Pycharm question better than I can.   *smiles*

dorodnic, 2019-02-24:
MobileNetSSD actually outputs multiple detections. In [distance_to_object.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) we intentionally take just the first one. Take a look at the content of `detections = net.forward("detection_out")`.
I'm closing this issue because this is not strictly related to RealSense. We will try to add "multi-object" example to SDK samples in the future to help make things easier.

closed 2019-02-24 by dorodnic