realsenseai/librealsense #3087: Working with Depth image

MaryBD, 2019-01-17:
Is there any example on working with depth frame, such as how to find the distance of the closest object to the camera, or how to obtain the depth value of the specific pixle in rgb frame. Thanks.

dorodnic, 2019-01-17:
Hi @MaryBD 
Did you try the [Distance to Object example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples#interactive-examples)?

MaryBD, 2019-01-17:
Yes I tried that. I just wonder if there are some functions to help me with finding the distance of the closest object to the camera ,.... without writing a loop and caaling each pixel one by one.

RealSenseCustomerSupport, 2019-01-22:
@MaryBD I'm afraid that you need do some loop and calculation to get the closest. Please have a try. Thanks!

RealSenseCustomerSupport, 2019-01-25:
@MaryBD Any other questions about this issue? Looking forward to your update. Thanks!

MaryBD, 2019-01-28:
I did a conventional loop and calculation to solve it.

closed 2019-01-28 by MaryBD