realsenseai/librealsense #3088: D435 - Crash when setting EnableAutoExposure (C#/UnityWrapper)

easchimidt, 2019-01-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello guys,

Code and error below. Do you guys knows how can I handle this kind of error (when setting options), avoiding the crash?

Thanks.

Code (C#):

` 
                Pipe = new Pipeline();
                Config = new Config();
                
                Config.EnableStream(Stream.Color, 1280, 720, Format.Rgba8, 30);
                Config.EnableStream(Stream.Infrared, 1280, 720, Format.Y8, 30);

                PipelineProfile profile = Pipe.Start(Config);

                Sensor sensorIR = profile.Device.Sensors[0];
                Sensor sensorColor = profile.Device.Sensors[1];

                foreach (var opt in sensorIR.Options)
                {
                    if (opt.Supported)
                    {
                        if (opt.Key == Option.EmitterEnabled)
                        {
                            opt.Value = 0f;
                        }
                        if (opt.Key == Option.EnableAutoExposure)
                        {
                            opt.Value = 1f; // -------> Error happens at this line
                        }
                    }
                }
`

Error:
Exception: get_xu(id=11) failed! Last Error: No error
rs2_get_option(options:0000000040D7D650, option:Enable Auto Exposure)
Intel.RealSense.Helpers+ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <4fd4dfd1857e4955a500c05c5081d4c2>:0)
Intel.RealSense.Sensor+CameraOption.get_Value () (at <4fd4dfd1857e4955a500c05c5081d4c2>:0)

easchimidt, 2019-01-17:
Just one fact that might not be related. When I try to use RealSense Viewer after this problem happens, the Viewer gets "stuck": I can't access anything (Add Source, On/Off cameras, 2D/3D, nothing). Don't know if it was a consequence of the code, or the camera was in a weird state before (note: it was the first usage after booting computer, so the camera wasn't being used before).

RealSenseCustomerSupport, 2019-01-22:
Hi @easchimidt,

What's your D435 firmware version? Can you upgrade it to the latest version.
I don't observe this issue based on a quick modified test based on the C# example "cs-tutorial-2-capture".

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi @easchimidt,

Any update? Do you still observe this issue in the latest librealsense?

easchimidt, 2019-03-05:
Hi. Sorry for the late reply. I've tried with the latest firmware, and the bug isn't happening anymore when disabling the emmiter. 

Thanks for the help!

closed 2019-03-14 by RealSenseCustomerSupport