realsenseai/librealsense #3092: T260 Availability

JBBee, 2019-01-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.1                          | 
| Language                            |  unity                          | 
| Segment			|  AR                          | 

### Issue Description
A few issues have asked about RS2_STREAM_POSE:    #1078 #3070 and as I've read it's a placeholder for T260 support.  Is the T260 tracking module with the Realsense SLAM vision library available, or when will it be available, for purchase?  

Thanks.

dorodnic, 2019-01-17:
Hi @JBBee 
Appreciate your interest in this product. T26x is not publicly available **yet** but please stay tuned at [realsense.intel.com](https://realsense.intel.com)

dorodnic, 2019-01-23:
More information on the T265 is now available (#3129)

closed 2019-01-23 by dorodnic