realsenseai/librealsense #3093: Running realsense viewer after installation

pillaya21, 2019-01-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (5.11.1.0) | 
| Operating System & Version |   {Linux 16.04} | 
| Kernel Version (Linux Only)    |  (4.15.0-43-generic) | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.17.1}                          | 

### Issue Description
I followed the installation instructions from here 
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
I didn't get any errors but when I tried to check the camera with "realsense-viewer" 
I got the following error. 

$ realsense-viewer 
 17/01 17:04:38,239 WARNING [140538471152256] (types.cpp:57) API version mismatch: librealsense.so was compiled with API version 2.11.1 but the application was compiled with 2.17.1! Make sure correct version of the library is installed (make install)
 17/01 17:04:38,239 WARNING [140538471152256] (rs.cpp:204) null pointer passed for argument "context"
RealSense error calling rs2_create_context(api_version:21701):
    API version mismatch: librealsense.so was compiled with API version 2.11.1 but the application was compiled with 2.17.1! Make sure correct version of the library is installed (make install)

How do I get it to work?

pillaya21, 2019-01-28:
Hello @ev-mp 

I got the realsense-viewer to work, but its not detecting the D435 sensor. 
[image]
The sensor is connected using a USB 3.0 
$ lsusb
Bus 002 Device 002: ID 8086:0b07 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0a2a Intel Corp. 
Bus 001 Device 003: ID 05c8:03ab Cheng Uei Precision Industry Co., Ltd (Foxlink) 
Bus 001 Device 002: ID 1267:0213 Logic3 / SpectraVideo plc 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The firmware is also updated.

RealSenseCustomerSupport, 2019-02-01:
----------------------------------------------

@pillaya21 Did you try v2.17.1 to see if any such warnings? These warnings should be caused by one problem of YUYV mode RGB streaming with v2.18.  If you still want to use v2.18, please choose other formats for RGB streaming and ignore these warnings. Or you can switch to try v2.17.1 Thanks!

closed 2019-02-01 by pillaya21