realsenseai/librealsense #3094: Realsense viewer Error: Unknown in rs2_create_device....

titusvp, 2019-01-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|----------------------- ---------|------------------------------------------- |
| Camera Model                       |  D435 } | 
| Firmware Version                   | (no info button in realsense viewer) | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC  |
| SDK Version                           |  { legacy / 2.<1>.<7> }                          | 
| Language                               |  {C++ }                          |                        | 

When starting the Realsense-Viewer i get the error "Unknown in rs2_create_device(info_list:0000019B47E72570, index:0):
Could not initialize extensions controls!"

It only appears on my laptop. i checked for other pc's with the same and different windows versions which both worked fine without error after only installing the sdk2.exe.

I also checked my privacy settings under windows>options>Privacy>Camera and theyre "on". Which was suggested in #1930.

Then i tried deinstalling the camera drivers in the device manager and reinstalled them by plugging the camera back into the laptop but the error still appears. (regarding #1179)

RealSenseCustomerSupport, 2019-01-22:
@titusvp  What's your RealSenseViewer version? Did you try v2.17.0 directly downloaded from https://github.com/IntelRealSense/librealsense/releases/download/v2.17.0/Intel.RealSense.Viewer.exe ?

RealSenseCustomerSupport, 2019-01-23:
@titusvp  Please refer to https://github.com/IntelRealSense/librealsense/issues/1930 to resolve the error. Thanks!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:02 PM

@titusvp Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-02-01:
----------------------------------------------

@titusvp Any update?