realsenseai/librealsense #3095: Cant run capture example via terminal compiler function

titusvp, 2019-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400} | 
| Firmware Version                   |  | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15. 0  )                                     | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.7.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  }                          | 

### Issue Description
When compiling the capture example via: g++ -o rs-capture rs-capture.cpp 
i get the following error:titus@titus-Latitude-7490:~/Desktop/test/librealsense/examples/capture$ g++ -o rs-capture rs-capture.cpp

In file included from /usr/include/c++/5/chrono:35:0,
                 from /usr/local/include/librealsense2/hpp/rs_types.hpp:23,
                 from /usr/local/include/librealsense2/rs.hpp:8,
                 from rs-capture.cpp:4:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
rs-capture.cpp:5:93: fatal error: example.hpp: No such file or directory
compilation terminated.

Did i mess up the set up or why isnt example.hpp pathed by cmake?
I followed the instructions in doc > installation.md

Alpatzino, 2019-01-18:
Had the same problem. It seams the file doesn't get placed properly during installation.

Copy the example.hpp from the directory to /usr/local/include/librealsense2 and set the path in rs-capture.cpp (or for whatever cpp you will use)

titusvp, 2019-01-18:
Thanks for your help Alpatzino.
I tried to follow your suggestion but its still not found. I copied the example.hpp to "/home/titus/Desktop/test/librealsense/include" and replaced the #include example.hpp in the capture.cpp though #include "/home/titus/Desktop/collisioncontrol/librealsense/include/example.hpp"

I can actually make the examples work by just executing them over the terminal via cd"path" and calling the .cpp. 
When trying to compile the file though, i get the error that example.hpp is not pathed. How does this make sense?

Alpatzino, 2019-01-18:
Replace the the #include example.hpp with #include </home/titus/Desktop/test/librealsense/include/example.hpp>. Atm you're setting the path to collisionscontrol etc. Can also work if you place the file there.

I'm not sure, I think it has sth. to do with how it searches differently if youre using terminal.

RealSenseCustomerSupport, 2019-01-23:
@titusvp  Any update about this issue? Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

XiangyuX Sun, Jan 28, 10:02 PM

@titusvp Any other questions about this ticket? If no, will close the ticket accordingly. Thanks!

closed 2019-02-01 by titusvp