realsenseai/librealsense #3096: Is your kabsch_rmsd calculation wrong?

honghande, 2019-01-18:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/calibration_kabsch.py
I looked at the principle of the Kabsch algorithm and found that your RMSD calculation is not correct: you only calculate the rotation matrix, but not the translation vector, and then calculate the RMSD. I don't know why.

dorodnic, 2019-01-18:
Hi @honghande 
Box Dimentioner example is contributed by the community. I tested that it works and does what it claims, but otherwise I don't think anyone on the team can comment on the algorithm. You are welcomed to contribute enhancements if you get better results.

honghande, 2019-01-19:
hi，@dorodnic
I just want to know how that algorithm works in the code and why it is not difficult for you, so help explain it, or you can help provide the author's origin, thank you!

dorodnic, 2019-02-24:
All code history is public on github, you are welcomed to reach out to the authors. 
However, the example is trying to demonstrate how to work with the camera and not how to correctly implement any particular algorithm.

closed 2019-02-24 by dorodnic