realsenseai/librealsense #3097: How to get depth from depth image without using depth frame

shadowdk3, 2019-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   Ubuntu16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-42-generic                                       | 
| Platform                                 | PC|
| SDK Version                            | 2.17.1                     | 
| Language                            |  python                          | 
| Segment			| Robot                     | 

### Issue Description
Hi, 
I want to get the distance of a point, if I use the aligned image that provided in github, there are many holds and the converted depth pixel  to metric value will become 0, 0, 0. So that I try to inpainting on depth image and get depth value from depth image convert to distance.

I found that the output in `get_depth_at_pixel` function is float. Now, I can get value from depth image, which is in `numpy.uint16`, how can I convert to `double/float` value? 

Thanks

chaojizhangdaxin, 2019-01-25:
我们获取的深度值就是以米为单位的实际距离，比如2.1212312312312312米

closed 2019-01-29 by shadowdk3

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Thank you @chaojizhangdaxin

Translating your comment:
"The depth value we get is the actual distance in meters, such as 2.1212312312312312 meters."

We were also unable to replicate issue, no comments over 20 days.

Close issue.